viesturz/klipper-toolchanger #4: Feature idea: Modifying BED_MESH_OFFSET

fiferboy, 2024-01-12:
Hi @viesturz - let me know if I am way off the mark with my thinking here.

Would it be beneficial to change BED_MESH_OFFSET in Klipper to allow Z offset adjustments like it does for X and Y? I understand the reasoning it wasn't built this way, but with multiple toolhead printers (toolchangers and even IDEX printers) it seems like it would be advantageous to control Z offset between toolheads the same way we do with X and Y and leave the live stepping/baby stepping for actually live tuning a print.

I am thinking of writing what is essentially a copy of the current BED_MESH_OFFSET module and adding in Z and adapting the toolchanger modules to use the new method as a proof-of-concept but I thought I would check with you first to see if you already looked down that road but found a reason it wouldn't work.

fiferboy, 2024-02-07:
A very similar PR was recently opened on the Klipper GitHub page - see https://github.com/Klipper3d/klipper/pull/6485

This appears to do exactly what I am after, so I will look into testing it. I believe I will need to rework a bit of the klipper-toolchanger code to utilize this, but maybe it can be hacked in with macros for testing. Any input, @viesturz ?

TheSin-, 2024-02-19:
it has been merged BTW

viesturz, 2024-04-03:
Hmm, I'm not fully following here.

BED_MESH_OFFSET is meant to re-align the bed mesh with nozzle offset.
Nothing to do with the baby stepping offsets on the fly.
My personal 2 cents- it's a kludge that should have never existed, and bed mesh should just recognize normal Gcode offsets.

I have considered to have a separate offset system on top of GCODE_OFFSET for the tool offsets, but decided against it to keep the initial version simple.

viesturz, 2024-04-03:
And updated toolchanger to take advantage of ZFADE, thanks!

VIN-y, 2024-06-16:
If it have been updated / merged. Shall we close this issue?