viesturz/klipper-toolchanger #7: TypeError: unsupported operand type(s) for +=: 'float' and 'NoneType'

ChGeorgi, 2024-01-23:
I am getting this for a Toolchange

Traceback (most recent call last):
  File "/home/g/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/g/klipper/klippy/extras/tool.py", line 102, in <lambda>
    func = lambda gcmd: tc.select_tool(
  File "/home/g/klipper/klippy/extras/toolchanger.py", line 236, in select_tool
    'restore_position': self._restore_position_with_tool_offset(
  File "/home/g/klipper/klippy/extras/toolchanger.py", line 342, in _restore_position_with_tool_offset
    v += offset
TypeError: unsupported operand type(s) for +=: 'float' and 'NoneType'
Transition to shutdown state: Internal error on command:"T0"

viesturz, 2024-01-23:
Looks like you do not have gcode_<x,y,z>_offset defined for the tool.

I pushed an update to default them to 0, not none. Let me know if that helps.
Otherwise please attach the klippy.log.

ChGeorgi, 2024-01-23:
omg,  you are right
i missed T0 because i used that as reference
[tool T1]
gcode_x_offset: 0
gcode_y_offset: 0
gcode_z_offset: 0
# -0.2

[tool T2]
gcode_x_offset: 0
gcode_y_offset: 0
gcode_z_offset:0

[tool T3]
gcode_x_offset: 0
gcode_y_offset: 0
gcode_z_offset: -1.2

closed 2024-04-03 by viesturz