viesturz/klipper-toolchanger #9: mcu crashing mid print

Marshalldog, 2024-01-27:
been having a weird issue and I think it may be related to the new updates. 
[image]
I previously had no issues and but since last week (or two weeks ago (not sure exactly)) the printer repeatedly crash mid print.
[moonraker (7).log](https://github.com/viesturz/klipper-toolchanger/files/14071321/moonraker.7.log)
[klippy (7).log](https://github.com/viesturz/klipper-toolchanger/files/14071322/klippy.7.log)
[crowsnest (4).log](https://github.com/viesturz/klipper-toolchanger/files/14071323/crowsnest.4.log)

at the time of the crash the load spiked and then crashed (note the secomd toolhead was idle)
mainboard
[image]
active toolhead board
[image]
inactive toolhead board
[image]

Marshalldog, 2024-01-27:
[image]
printer.cfg
```

[include mainsail.cfg] #fixed for my setup
[include timelapse.cfg]
[exclude_object]

[include mcu.cfg]
#[include adxlmcu.cfg]

#my shit
[include FilaSenManagement.cfg]
#[include LEDManagement.cfg]

#[include KAMP/Adaptive_Meshing.cfg]
#[include KAMP/Line_Purge.cfg]] 

#[include leds.cfg]

[include tool_detection.cfg]
[include toolchanger.cfg]

[include T0-burner-dragon.cfg]
[include T1-Rapid-Burner.cfg]

[include homing.cfg]
[include calibrate-offsets.cfg]
[include macros.cfg]

#fix shit
## Add virtual_sdcard
[virtual_sdcard]
path: /home/pi/printer_data/gcodes
on_error_gcode: CANCEL_PRINT

### Add support for respond
[respond]

### Add support for pause_resume
[pause_resume]

### Add support for save variables
[save_variables]
filename:  ~/variables.cfg
[gcode_arcs]
#resolution: 1.0
#   An arc will be split into segments. Each segment's length will
#   equal the resolution in mm set above. Lower values will produce a
#   finer arc, but also more work for your machine. Arcs smaller than
#   the configured value will become straight lines. The default is
#   1mm.
resolution: 0.2

[rounded_path]
resolution: 0.2
#replace_g0: True
replace_g0: False

[printer] #fixed for my setup
kinematics: corexy
max_velocity: 300  
max_accel: 3000             #Max 4000
max_z_velocity: 15          #Max 15 for 12V TMC Drivers, can increase for 24V
max_z_accel: 350
square_corner_velocity: 5.0

[force_move] #fixed for my setup
enable_force_move: True 

[... 812 lines cut ...]
      {% set newparameters = newparameters ~ " TARGET="~params.S %}
    {% endif %}
    SET_TOOL_TEMPERATURE{newparameters}
  {% else %}
    M104.1 {rawparams}
  {% endif %}

[gcode_macro M109]
rename_existing: M109.1
description: [T<index>] [S<temperature>]
  Set tool temperature and wait.
  T= Tool number, optional. If this parameter is not provided, the current tool is used.
  S= Target temperature
gcode:
  {% if params.T is defined %}
    {% set newparameters = " T="~params.T %}
    {% if params.S is defined %}
      {% set newparameters = newparameters ~ " TARGET="~params.S %}
    {% endif %}
    SET_TOOL_TEMPERATURE WAIT=1 {newparameters}
  {% else %}
    M109.1 {rawparams}
  {% endif %}

```

Marshalldog, 2024-01-27:
i even tried a completely new os image 
i installed 32bit lite, then used Kiauh to install the essentials and then installed klipper tool changer and this error still occurs mid print

Marshalldog, 2024-01-28:
i've done a ton of testing on the canbus stuff and I'm pretty sure its not canbus

Marshalldog, 2024-01-28:
The thing is I can replicate this on another Pi (with a fresh install) with 100% reliability

Surion79, 2024-02-18:
do you use both toolheads? maybe try it with one single Toolhead and then the other toolhead only to see if it is an issue with a toolhead? just an idea about process of elimination

closed 2024-06-11 by viesturz