viesturz/klipper-toolchanger #13: Klipper crash when SELECT_TOOL is done before home

VIN-y, 2024-04-17:
When the user accidentally click on the  macro `T0`, `T1`, etc. before the axis are homed, Klipper will crash.

My work-around so far is to modify the macro as such:

```
[gcode_macro T0]
variable_color: ""
gcode:
    # SELECT_TOOL T=0
    {% if "xyz" in printer.toolhead.homed_axes %}    ### Check if axes are homed
        SELECT_TOOL T=0
    {% else %}
        RESPOND TYPE=error MSG='Printer is not homed'
    {% endif %}
```

It works, but this generate a bunch of repetitive codes (as we discussed in the tapchanger PR, a while back), as the that "home check" need to be copy into every new `Tx` macros. Therefore, it would be nice if the `SELECT_TOOL` comnand would check for the home-status before the `gcode_position` variable is set/used, line 259 in [toolchanger.py](https://github.com/viesturz/klipper-toolchanger/blob/main/klipper/extras/toolchanger.py).

viesturz, 2024-04-18:
Thanks, adding automatic lazy homing.

VIN-y, 2024-04-18:
I would suggest just return an error message “Printer is not homed”. In the same theme of accidental touches, it is best that the code at this level (i.e. where most people will never look-at/edit) does not assume what people would like to the motion system to do. 

If people want to automatically home, if it is not already so, when `Tx` macro is pressed, then they can mod the macro for that.

viesturz, 2024-04-20:
Added both options at the end. 
Personally like to minimize script writing. And home and tool buttons are equally dangerous.

https://github.com/viesturz/klipper-toolchanger/commit/3b4c11f6e2f3e0577067b1334ec8fdd1f0c2c1e7#diff-f204edcb110fd75353cfcbe6806701536306ce5a47a4cfbddba7a64fac8bc512

closed 2024-04-20 by viesturz