viesturz/klipper-toolchanger #20: Update issue! get_probe_params

JuniorGamingTime, 2024-06-11:
https://github.com/viesturz/klipper-toolchanger/blob/83873f82a9a64de35cafe698965efa2ef51f7adc/klipper/extras/tool_probe.py#L95

There seems to be an error when doing Quad Gantry Leveling and Bed Mesh Calibrate.
There's a pop up error that says:
"get_probe_params() takes from 1 to 2 positional arguments but 3 was given"

[image]

And here is my homing.cfg
`
#########################################################################################
# HOMING #
#########################################################################################

[homing_override]
axes: xyz
gcode:
    ## Trigger variables
    {% set x = 0 %}
    {% set y = 0 %}
    {% set z = 0 %}
    {% set a = 0 %}

    ## Set up variables
    {% set center_x = 175 %}
    {% set center_y = 175 %}

    ##  Engage trigger variables according to the input
    {% if params.X is defined %}
        {% set x = 1 %}                                                    # for x home
    {% endif %}
    {% if params.Y is defined %}
        {% set y = 1 %}                                                    # for y home
    {% endif %}
    {% if params.Z is defined %}
        {% set z = 1 %}                                                    # for z home
    {% endif %}
    {% if (x == 0 and y == 0 and z == 0) %}
        {% set a = 1 %}                                                    # for x home
    {% endif %}

    ### Homing Actions
    # Mandatory clearance moves ------------------------------------------------------
    G91                                                                        # Relative mode
    {% if x == 1 or y == 1 or a == 1 %}
        {% if "z" in printer.toolhead.homed_axes %}                            # If z is already homed
            {% set cur_z = printer.toolhead.position[2]|float %}
            {% if cur_z < 15.0 %}
                RESPOND TYPE=echo MSG='Nozzle Clearance...'
                G1 Z{ 15 - cur_z } F1500                                       # move Z up
            {% endif %}
        {% else %}                                                             # but if x is not homed
            {% if  printer.tool_probe_endstop.active_tool_number|int == -1 %}
                RESPOND TYPE=echo MSG='Nozzle Clearance...'
                SET_KINEMATIC_POSITION Z=0                                     # Set curent Z position as 0
                G1 Z+15 F3000                                                  # move Z up
                {% if a != 1 %}
                    M84                                                        # remove the temporary postions
                {% endif %}
            {% endif %}
        {% endif %}
            
        INITIALIZE_TOOLCHANGER
        STOP_TOOL_PROBE_CRASH_DETECTION
        
        {% if "z" in printer.toolhead.homed_axes %}                            # If z is already homed
            {% set cur_z = printer.toolhead.position[2]|float %}
            {% if cur_z > 200.0 %}
                RESPOND TYPE=echo MSG='Umbilical Clearance...'
                G1 Z{ 200.0 - cur_z } F3000                                    # move Z down
            {% endif %}
        {% else %}
            RESPOND TYPE=echo MSG='Umbilical Clearance...'
            G28 Z
            G1 Z+15 F3000                                                      # Move Z up off the bed
        {% endif %}
        G90                                                                    # Asolute mode
        
        # Specific opition moves ------------------------------------------------------------
        {% if x == 1 or y == 1 or a == 1 %}                                    # If X or Y is called
            RESPOND TYPE=echo MSG='Home Y...'
            G28 Y                                                              # Home Y
        {% endif %}    
        
        {% if x == 1 or a == 1 %}                                              # If X is called
            RESPOND TYPE=echo MSG='Home X...'
            G28 X                                                              # Home X
        {% endif %}
    
        {% if z == 1 or a == 1 %}                                              # If "G28" or "G28 Z" is called
            RESPOND TYPE=echo MSG='Home Z...'
            G1 X{center_x} Y{center_y} F10000                                  # Move the toolhead to the centre of the bed
            G28 Z                                                              # Home Z
            _ADJUST_Z_HOME_FOR_TOOL_OFFSET                                     # Apply toolhead specific Z-offset
        {% endif %}
        _APPLY_ACTIVE_TOOL_GCODE_OFFSETS                                       # Apply the X and Y toolhead offsets
        M400
    
    {% else %}
        RESPOND TYPE=echo MSG='Home Z...'
        {% set cur_z = printer.toolhead.position[2]|float %}
        {% if cur_z < 15.0 %}
            G1 Z{ 15 - cur_z } F1500                                       # move Z up
        {% endif %}
        {% if "xyz" in printer.toolhead.homed_axes %}
            G90
            G1 X{center_x} Y{center_y} F10000
        {% endif %}
        G28 Z                                                              # Home Z
        _ADJUST_Z_HOME_FOR_TOOL_OFFSET                                     # Apply toolhead specific Z-offset
    {% endif %}

#########################################################################################
# QUAD GANTRY LEVELING # 
#########################################################################################
[quad_gantry_level]
gantry_corners:
   -60,-10
   410,420

points:
   30,10
   30,310
   330,310
   330,10
 

speed: 350
horizontal_move_z: 10
retries: 5
retry_tolerance: 0.025
max_adjust: 50

#########################################################################################

; Depending on the selected tool at the time of homing, the physical Z endstop position is offset.
; This corrects for that using current tool offset.
[gcode_macro _ADJUST_Z_HOME_FOR_TOOL_OFFSET]
gcode:
      G90 ; absolute mode
      G0 Z10 F1000
      {% set tool = printer.toolchanger.tool %}
      {% if tool %}
         {% set tool_z_offset = printer[tool].gcode_z_offset %}
         {% set probe_z_offset = printer.tool_probe_endstop.active_tool_probe_z_offset %}
         SET_KINEMATIC_POSITION Z={10.0+tool_z_offset|float+probe_z_offset|float}
      {% endif %}

[gcode_macro _APPLY_ACTIVE_TOOL_GCODE_OFFSETS]
gcode:
    ; Apply gcode offsets
    {% set tool = printer.toolchanger.tool %}
    {% if tool %}
      SET_GCODE_OFFSET X={printer[tool].gcode_x_offset} Y={printer[tool].gcode_y_offset} Z={printer[tool].gcode_z_offset}
    {% endif %}

[gcode_macro TOOL_BED_MESH_CALIBRATE]
gcode:
      {% set tool_z_offset = printer[printer.toolchanger.tool].gcode_z_offset %}
      G90 ; absolute mode
      G0 Z10 F1000
      # Bed mesh knows about the probe offset, but not about the tool offset.
      SET_KINEMATIC_POSITION Z={10.0-tool_z_offset|float}
      BED_MESH_CALIBRATE
      G0 Z10 F1000
      SET_KINEMATIC_POSITION Z={10.0+tool_z_offset|float}`

RNGIllSkillz, 2024-06-12:
Fixed this error in the latest pull request

JuniorGamingTime, 2024-06-12:
Latest update fixed the errors! Thanks guys 💪🫡

closed 2024-06-12 by JuniorGamingTime