viesturz/klipper-toolchanger #32: Dissable tool_probe

zi-vo, 2024-09-13:
Hi, 
i try to use the klipper-toolchanger extension for a self-build toolchanger approach. Due to a different setup, i dont use the tool-probe as the tap changer does. This leads to multiple errors. First of all, i can't use the tool_detection.cfg as this creates a duplicate of the probe pin. 
Currently there is no way of changing the tool_detection behavior which shuts down any approach of using this without a voron tap system.

Not including the tool-detection file leads to an error after the tool pickup, cause no detection can be done. If there would be a way to implement own tool detection routines, this would be really nice

Rneil-ca, 2024-12-14:
Where you ever able to figure this one out?  I spent a bit a time trying to disable the tool_prope extension, but it is so integral to the rest of the extensions, that I gave up.  Would really like to get this working.  Ended up using the KTC repository from TypQxq and have gotten that working, but like some of the options of this repository.

Thank you viesturz for all your hard work on this and the tap changer repository.  I am working on adapting them to my Vzbot.
Ryan

viesturz, 2024-12-16:
The tool probe is completely optional.
You would need to delete a bunch of calls from the macros and somehow
specify the active tool on initialization.

I'm building a new toolchanger version with a probe on just a single tool,
and will  update the code to specify a presence detection pin for each
tool, making this easier.

viesturz, 2024-12-16:
This should do the trick:
https://github.com/viesturz/klipper-toolchanger/commit/572c2848c2d1bb51b782e2f748c6055e722ef349

Rneil-ca, 2024-12-18:
Wow, looks like you put a lot of work into this, thank you.  Will try and load it soon when I have some time.

Rneil-ca, 2025-07-29:
I was able to add the tool crash protection by modifying the toolchanger.py with some of the code used in the tool_probe side.  It works and has been tested, might need tweaking to add it permanently.  File added as txt as it will not let you add py files to chat.

[toolchanger.txt](https://github.com/user-attachments/files/21479031/toolchanger.txt)

Thanks again for all your work on this.

Ryan