viesturz/klipper-toolchanger #33: Feature request : Filament detection sensor per toolhead

orotecluc, 2024-09-16:
In my setup for ToolChanger, I am planninng on : 

- either have an extrernal filamentsensor (type BTT motion sensor or so) )per toolhead just before the entry of filamnet, and connected to my MainBoard
- Or have a filaments sensor switch just at entry of extruder (like they have done with the ERFC2 project), and thus have a microswitch wil ball installed into the extruder housing, and connected to the EBB

So ideal, we would have to be able to assign any tipe of filament sensor (by name) to any tool(by Id)

I am not sure if this is feasible

matthiasheller69, 2025-01-29:
Hello, yes, I'm interested in that too. I have implemented a Voron 2.4 with 6-fold toolchanger. Optionally I would also like to implement a runout sensor for each tool head. However, I have no idea how this can be implemented in the depths of the code.

viesturz, 2025-03-07:
You can just configure a filament switch sensor for each toolead. They are not conflicting.
Extra points for adding SET_FILAMENT_SENSOR SENSOR=T0_sensor ENABLE=0/1 in the tool change macro.

Filament width sensor is quite a bit more complex beast, and I don't think you can practically install one for each tooleahd.

RainerBavaria, 2025-04-20:
> You can just configure a filament switch sensor for each toolead. They are not conflicting. Extra points for adding SET_FILAMENT_SENSOR SENSOR=T0_sensor ENABLE=0/1 in the tool change macro. [...]

what do this parse do ? where do I need to place it ?