viesturz/klipper-toolchanger #35: Feature Request - Handling multiple X Endstops, one on each tool

robwaldhauser, 2024-12-22:
klipper-toolchanger has been excellent to use, however there is one scenario where the answer isn't clear or possibly not implemented.

Our Z probe is also our X probe, meaning that our X endstop is wired inline with our z probe for each tool. This means we can use z_virtual_endstop for our Z endstop, but Klipper will bark when attempting to use for X as well. We can use duplicate pin override to utilize 1 of the tools x endstops, however we have not found a way to use any of them based on which tool is connected. 

Is this possible currently, or is there development that could make this happen? It would really help to allow for some great toolchanging designs that are being worked on! Thanks!

viesturz, 2024-12-22:
Hmm, interesting use case. [tool_probe](./tool_probe.md) does this, but for Z only.

This is *almost* possible today by just using:
```
[stepper_y]
endstop_pin: probe:z_virtual_endstop
```

But it overrides the endstop position (at 0+probe offset) which is not generally what you want.
It should be fairly easy to change it around, let me see if I can make it work.

robwaldhauser, 2024-12-22:
> Hmm, interesting use case. [tool_probe](./tool_probe.md) does this, but for Z only.
>
> This is _almost_ possible today by just using:
>
> ```
> [stepper_y]
> endstop_pin: probe:z_virtual_endstop
> ```
>
> But it [...]

Specifically we are hoping for X to be tied to the same pins as z_virtual_endstop, but if we add that Klipper will bark that we can't use it for both our X endstop and our Z endstop. If you think this can pretty easily to implement that would be incredible and I'd love to test if you need any confirmation that it works!

viesturz, 2024-12-22:
Added something in a branch, see https://github.com/viesturz/klipper-toolchanger/blob/xy-endstop-routing/tool_probe.md

Very not tested, may burn your printer down, etc, etc. This is somewhat hacky and I would like to rewrite at some point to allow arbitrary pin routing based on selected tool.

robwaldhauser, 2024-12-22:
That was fast! I'm not home currently but will test and play around with it as soon as im home tonight and will report back (hardware is already set up to go). Thanks so much!

robwaldhauser, 2024-12-23:
> Added something in a branch, see https://github.com/viesturz/klipper-toolchanger/blob/xy-endstop-routing/tool_probe.md
> 
> Very not tested, may burn your printer down, etc, etc. This is somewhat hacky and I would like to rewrite at some point to allow arbitrary pin routing based on selected tool.

I tried this out and essentially swapped out the single pin number in my x axis endprobe to what you have in your instructions. While triggering it manually, it does show correct when I query endstops. However when I do a homing function, it treats X axis as if it were using sensorless homing.

Hope that makes sense, happy to share anything else if there's something that can help you pinpoint that.

runningsystemchanger, 2025-01-04:
I almost got it to work, it's homing nicely (X sensored, Y sensorless, then Z sensored) but after Z home I get:
Endstop z still triggered after retract
Endstop z still triggered after retract
Endstop z still triggered after retract
Of course followed by the klipper shutdown.

kensand, 2025-02-08:
Also very appreciative of your hard work @viesturz 😃 

Just to clarify usage of this, the GPIO pin that is wired to my z-probe needs to also be wired to my x-endstop? Presumably ground and +5v for my z-probe and +5v for my endstop don't need to be wired together though, correct? Then in my config file, I simply replace the x-endstop pin with `tool_probe_endstop:x_virtual_endstop`?

Thanks!

printicus, 2025-03-10:
I've got, kinda, the same problem... I want x endstop on every toolhead. Because I use classic microswitches for x enstop, each with his personal gpio from the toolboard, I need to tell the printer that the x endstop can be triggered from multiple pins... Is that possible?

kensand, 2025-04-12:
Finally got some time to try this. Connected the power and signal pins between my x-endstop switch and z-probe, and I am also getting the same behavior as @noobydp: `Endstop x still triggered after retract` on the second triggering of the switch. I'll see if I can't debug further at least, now that I have a toolhead setup to test this with.

kensand, 2025-04-12:
I managed to get this working a different way by modifying the [multi_pin](https://www.klipper3d.org/Config_Reference.html#multi_pin) configuration that ships with Klipper itself to allow using it for endstops. This has the advantage of not requiring the z-probe and x-endstop to be wired together (which I find a bit cleaner and simpler). I will clean it up, make it into a new config type, and make a PR either here or to Klipper (@viesturz any preference?), but in the meantime, if anyone is dying to try this themselves, here is the new version of the multi_pin.py that I put together: 

```python
class PrinterMultiPin:
    def __init__(self, config):
        self.printer = config.get_printer()
        ppins = self.printer.lookup_object('pins')
        try:
            ppins.register_chip('multi_pin', self)
        except ppins.error:
            pass
        self.pin_type = None
        self.pin_list = config.getlist('pins')
        self.mcu_pins = []
        self.steppers = []

    def setup_pin(self, pin_type, pin_params):
        ppins = self.printer.lookup_object('pins')
        pin_name = pin_params['pin']
        pin = self.printer.lookup_object('multi_pin ' + pin_name, None)
        if pin is not self:
            if pin is None:
                raise ppins.error("multi_pin %s not configured" % (pin_name,))
            return pin.setup_pin(pin_type, pin_params)
        if self.pin_type is not None:
            raise ppins.error("Can't setup multi_pin %s twice" % (pin_name,))
        self.pin_type = pin_type
        invert = ""
        if pin_params['invert']:
            invert = "!"
        pullup = ""
        if pin_params['pullup']:
            pullup = "^"
        self.mcu_pins = [ppins.setup_pin(pin_type, invert + pullup + pin_desc)
                         for pin_desc in self.pin_list]
        return self

    def get_mcu(self):
        return self.mcu_pins[0].get_mcu()

    def setup_max_duration(self, max_duration):
        for mcu_pin in self.mcu_pins:
            mcu_pin.setup_max_duration(max_duration)

    def setup_start_value(self, start_value, shutdown_value):
        for mcu_pin in self.mcu_pins:
            mcu_pin.setup_start_value(start_value, shutdown_value)

    def setup_cycle_time(self, cycle_time, hardware_pwm=False):
        for mcu_pin in self.mcu_pins:
            mcu_pin.setup_cycle_time(cycle_time, hardware_pwm)

    def set_digital(self, print_time, value):
        for mcu_pin in self.mcu_pins:
            mcu_pin.set_digital(print_time, value)

    def set_pwm(self, print_time, value):
        for mcu_pin in self.mcu_pins:
            mcu_pin.set_pwm(print_time, value)

    def add_stepper(self, stepper):
        self.steppers.append(stepper)

    def get_steppers(self):
        return self.steppers

    def query_endstop(self, print_time):
        res = False
        for mcu_pin in self.mcu_pins:
            res = res or mcu_pin.query_endstop(print_time)
        return res

    def home_start(self, print_time, endstop_sample_time,
                   endstop_sample_count, rest_time,
                   triggered):
        return self.multi_complete([
            mcu_pin.home_start(
                print_time,
                endstop_sample_time,
                endstop_sample_count,
                rest_time,
                triggered
            ) for mcu_pin in self.mcu_pins], 10)

    def home_wait(self, home_end_time):
        arr = [mcu_pin.home_wait(home_end_time) for mcu_pin in self.mcu_pins]
        return max(arr)

    # Return a completion that completes when any completions in a list complete
    def multi_complete(self, completions, wait_time):
        if len(completions) == 1:
            return completions[0]
        # Build completion that waits for all completions
        reactor = self.printer.get_reactor()

        def wait_for_any(eventtime):
            while True:
                for c in completions:
                    res = c.wait(wait_time, None)
                    if res is not None and c.test():
                        return res

        return reactor.register_callback(wait_for_any)

def load_config_prefix(config):
    return PrinterMultiPin(config)

```

and this is how I'm using it in my config:

```
[multi_pin xendstop]
pins: ^Tool0:PB9,^Tool1:PB9,^Tool2:PB9,^Tool3:PB9,^Tool4:PB9,^Tool5:PB9

[stepper_x]
step_pin: PE6
dir_pin: PE5
enable_pin: !PC14
microsteps: 16
rotation_distance: 40
full_steps_per_rotation: 200  #set to 400 for 0.9 degree stepper
endstop_pin: multi_pin:xendstop
```

printicus, 2025-04-29:
> I managed to get this working a different way by modifying the [multi_pin](https://www.klipper3d.org/Config_Reference.html#multi_pin) configuration that ships with Klipper itself to allow using it for endstops. This has the advantage of not requiring the z-probe and x-endstop to be wired together (which I find a bit cleaner and simpler). I will clean it up, make it into a new config type, and make a PR either here or to Klipper ([@viesturz](https://github.com/viesturz) any preference?), but in the meantime, if anyone is dying to try this themselves, here is the new version of the [multi_pin.py](https://github.com/Klipper3d/klipper/blob/master/klippy/extras/multi_pin.py) that I put together:
> 
> class PrinterMultiPin:
>     def __init__(self, config):
>         self.printer = config.get_printer()
>         ppins = self.printer.lookup_object('pins')
>         try:
>             ppins.register_chip('multi_pin', self)
>         except ppins.error:
>             pass
>         self.pin_type = None
>         self.pin_list = config.getlist('pins')
>         self.mcu_pins = []
>         self.steppers = []
> 
>     def setup_pin(self, pin_type, pin_params):
>         ppins = self.printer.lookup_object('pins')
>         pin_name = pin_params['pin']
>         pin = self.printer.lookup_object('multi_pin ' + pin_name, None)
>         if pin is not self:
>             if pin is None:
>                 raise ppins.error("multi_pin %s not configured" % (pin_name,))
>             return pin.setup_pin(pin_type, pin_params)
>         if self.pin_type is not None:
>             raise ppins.error("Can't setup multi_pin %s twice" % (pin_name,))
>         self.pin_type = pin_type
>         invert = ""
>         if pin_params['invert']:
>             invert = "!"
>         pullup = ""
>         if pin_params['pullup']:
>             pullup = "^"
>         self.mcu_pins = [ppins.setup_pin(pin_type, invert + pullup + pin_desc)
>                          for pin_desc in self.pin_list]
>         return self
> 
>     def get_mcu(self):
>         return self.mcu_pins[0].get_mcu()
> 
>     def setup_max_duration(self, max_duration):
>         for mcu_pin in self.mcu_pins:
>             mcu_pin.setup_max_duration(max_duration)
> 
>     def setup_start_value(self, start_value, shutdown_value):
>         for mcu_pin in self.mcu_pins:
>             mcu_pin.setup_start_value(start_value, shutdown_value)
> 
>     def setup_cycle_time(self, cycle_time, hardware_pwm=False):
>         for mcu_pin in self.mcu_pins:
>             mcu_pin.setup_cycle_time(cycle_time, hardware_pwm)
> 
>     def set_digital(self, print_time, value):
>         for mcu_pin in self.mcu_pins:
>             mcu_pin.set_digital(print_time, value)
> 
>     def set_pwm(self, print_time, value):
>         for mcu_pin in self.mcu_pins:
>             mcu_pin.set_pwm(print_time, value)
> 
>     def add_stepper(self, stepper):
>         self.steppers.append(stepper)
> 
>     def get_steppers(self):
>         return self.steppers
> 
>     def query_endstop(self, print_time):
>         res = False
>         for mcu_pin in self.mcu_pins:
>             res = res or mcu_pin.query_endstop(print_time)
>         return res
> 
>     def home_start(self, print_time, endstop_sample_time,
>                    endstop_sample_count, rest_time,
>                    triggered):
>         return self.multi_complete([
>             mcu_pin.home_start(
>                 print_time,
>                 endstop_sample_time,
>                 endstop_sample_count,
>                 rest_time,
>                 triggered
>             ) for mcu_pin in self.mcu_pins], 10)
> 
>     def home_wait(self, home_end_time):
>         arr = [mcu_pin.home_wait(home_end_time) for mcu_pin in self.mcu_pins]
>         return max(arr)
> 
>     # Return a completion that completes when any completions in a list complete
>     def multi_complete(self, completions, wait_time):
>         if len(completions) == 1:
>             return completions[0]
>         # Build completion that waits for all completions
>         reactor = self.printer.get_reactor()
> 
>         def wait_for_any(eventtime):
>             while True:
>                 for c in completions:
>                     res = c.wait(wait_time, None)
>                     if res is not None and c.test():
>                         return res
> 
>         return reactor.register_callback(wait_for_any)
> 
> 
> def load_config_prefix(config):
>     return PrinterMultiPin(config)
> and this is how I'm using it in my config:
> 
> ```
> [multi_pin xendstop]
> pins: ^Tool0:PB9,^Tool1:PB9,^Tool2:PB9,^Tool3:PB9,^Tool4:PB9,^Tool5:PB9
> 
> [stepper_x]
> step_pin: PE6
> dir_pin: PE5
> enable_pin: !PC14
> microsteps: 16
> rotation_distance: 40
> full_steps_per_rotation: 200  #set to 400 for 0.9 degree stepper
> endstop_pin: multi_pin:xendstop
> ```

Tried this solution and, sadly, did not work for me.
This is how I used it:

`[multi_pin xendstop]
pins: nhk0:gpio13,nhk1:gpio13,nhk2:gpio13

## X Stepper on Motor1(B Motor)
[stepper_x]
step_pin: PE2
dir_pin: !PB4
enable_pin: !PC11
microsteps: 16
rotation_distance: 40
full_steps_per_rotation:200  #set to 400 for 0.9 degree stepper

endstop_pin: multi_pin:xendstop`