viesturz/klipper-toolchanger #37: tool_probe not applying the offset

Mamoth-42, 2025-01-07:
I have run into an issue with my tapchanger setup, Everything seems to be working aside from one thing: No matter which value i set in the [tool_probe] section under the "z_offset:" parameter, the applied offset always is zero. This applies to all toolheads, simply put, none of the offsets work. (I've tried renaming the probes and everything i could, no success) 

I've tried circumventing this issue by setting the gcode_z_offset under the [tool] section, however, this quickly turned out to not be suitable as my toolheads vary quite a bit in the nozzle-to-optoswitch length, therefore this way, homing with one toolhead will set the default gantry height considerably higher than homing with the other one, making succesful toolchanging impossible.

I've attached the printer.cfg in a txt form (github didn't like the .cfg)
[printer.txt](https://github.com/user-attachments/files/18338049/printer.txt)

Thank you in advance for any help and/or bugfixes

Mamoth-42, 2025-01-07:
It has just occured to me that a 
[klippy.log](https://github.com/user-attachments/files/18338096/klippy.log)
might be useful too

ga73, 2025-02-01:
I'm having exactly the same issue from a new install 3 days ago....no matter what the z_offset will not change even though it's set in the right place (tool probe z_offset field)

halfbro, 2025-02-06:
Check your console output after trying to do a Z home - if you have a `Unknown command:"SET_KINEMATIC_POSITION"`, then you will need to enable force moves as per the klipper docs [here](https://www.klipper3d.org/Config_Reference.html#force_move) and [here](https://www.klipper3d.org/G-Codes.html#set_kinematic_position)

cicinovec, 2025-02-23:
I have the same issue. For T0 the probe offset works only if placed in printer.cfg
For other tools it uses also the one from printer.cfg
No errors in the console

[klippy (6).log](https://github.com/user-attachments/files/18928710/klippy.6.log)

Mamoth-42, 2025-03-08:
@halfbro It would be nice if there were errors or something to begin with, but there's nothing like that. The printer thinks its working fine. Even though there's clearly a feature missing.

crbertilson, 2025-04-14:
I am having the same issue. Just built the machine this weekend. @cicinovec as a workaround are you copying the [tool_probe T0] to the printer.cfg or configuring a [probe]?

crbertilson, 2025-04-18:
> I am having the same issue. Just built the machine this weekend. [@cicinovec](https://github.com/cicinovec) as a workaround are you copying the [tool_probe T0] to the printer.cfg or configuring a [probe]?

I was able to resolve my issue.  I had a [homing_override] section in one of my old configs that was not allowing all the proper macros to run.  My offsets are now being applied properly.  I was not the original user to post this issue but wanted to share that my issue has been resolved.