viesturz/klipper-toolchanger #43: Stealthchanger path collision with dock

cicinovec, 2025-02-21:
Stealthchanger + modular dock + pickup path results in collision with dock.

When using the path, after picking the tool, the toolhead moves in straight line to safe_y.
But on the dock there are on each side notches the tool must avoid

Path used:
params_pickup_path: [{'z':-12, 'y':2}, {'z':-12, 'y':0}, {'z':1.5, 'y':0, 'f':0.5, 'verify':1}, {'z':0, 'y':0}, ]

Path updated:
params_pickup_path: [{'z':-12, 'y':2}, {'z':-12, 'y':0}, {'z':1.5, 'y':0, 'f':0.5, 'verify':1}, {'z':0.5, 'y':2.5, 'f':0.5}, {'z':8, 'y':8}, ]

[image]

viesturz, 2025-02-21:
Hi, thanks for the report. It' s been a while since I have used a stealthchanger.
One question before I update: for dropoff path does the above path in reverse work, or do you recommend a different one?

cicinovec, 2025-02-22:
Hi, dropoff also needed update, it hits the same notch.
I was tweaking the values for both docking and picking and those are currently functional values.
params_dropoff_path: [{'z':3.5, 'y':4}, {'z':0, 'y':0}, {'z':-12, 'y':0}]
params_pickup_path: [{'z':-12, 'y':2}, {'z':-12, 'y':0}, {'z':1.5, 'y':0, 'f':0.5, 'verify':1}, {'z':0.5, 'y':2.5, 'f':0.5}, {'z':8, 'y':8}, ]

closed 2025-03-09 by viesturz