viesturz/klipper-toolchanger #44: Feature request: Implement parent tool

piposa, 2025-02-26:
I try to use my toolchanger together with a MMU. I think a 2-tool toolchanger together with a MMU could be a "sweet spot" for effort. Change MMU color while the other parent tool is printing. I see in the README that the config options for that are already in it. But I don't think anything is implemented?!
Is there any near-term plan or anybody else working on this topic?

viesturz, 2025-03-06:
Thanks for reporting in, will take a look at this over the weekend.
It's been in the backlog, but no practical use so far :)

piposa, 2025-03-06:
I have a practical use and commit to test it asap :-) Right now I try to get it working using macros, but need to test more. Thanks a lot in any case!