viesturz/klipper-toolchanger #47: Fan speed not transfered if using SET_FAN_SPEED FAN=xyz

piposa, 2025-03-05:
With the latest commits "deprecate multi_fan and fold the fan management into [toolchanger]." the fan speed is not transfered if using SET_FAN_SPEED FAN=xyz. M106 Sx works as expected. Problem is relevant as Mainssail uses SET_FAN_SPEED FAN=xyz to set fan speed.
Test:
T0
SET_FAN_SPEED FAN=T0_partfan SPEED=100
T2
=> T0 fan speed is set to 0. T2 fan speed stays at 0

T0
M106 S100
T2
=> works as expected fanspeed T0 fan speed is 0 after change. T2 fan speed is set to 100

piposa, 2025-03-05:
Retested, now it works ... not sure what the issue was, sorry

piposa, 2025-03-05:
Issue is existing, but not totally clear what causes it, might be related to SELECT_TOOL T=x ? Not sure but at least transfer is not always working for me

viesturz, 2025-03-05:
Thanks for the report, will look into this.
Just to clarify - this is an issue with fan speed being set via the UI, not
the slicer.

And the expected behavior is to transfer the current fan speed to the next
tool?

One more thing that is not currently covered is select a fan via P{n} -
cura style.

piposa, 2025-03-06:
Correct, to test the behavior before aprint I was using the GUI and console. No actual printing so far. When I stopped testing the behavior was pretty random, which means I did not understand the root cause....

viesturz, 2025-03-06:
Hi, pushed an update to address this.
See https://github.com/viesturz/klipper-toolchanger/blob/main/toolchanger.md#gcodes-if-fan-is-specified-for-any-of-tools
And let me know if this works with mainsail.

closed 2025-03-06 by viesturz

piposa, 2025-03-06:
Not working!
Test: 
T0 selected
Use Mainsail => SET_FAN_SPEED FAN=T0_partfan SPEED=0.66
T2
Result: all fans off