viesturz/klipper-toolchanger #48: Bed Mesh not being loaded after toolchange, causing z to go lower after toolchange

0101falcon, 2025-03-06:
Using "per tool probe". When printing Klipper shows some z height, but actual z height is different (because of bed mesh), this causes z to negative when layer tool change happens: 
See console:
10:44
Selected tool 0 (tool T0)
10:44
Restore_pos_z 0.4
10:44
Current z 0.570684825965172 (This number is caused because of bed mesh)

(Check message.txt for Respond commands set by me in toolchanger.cfg.)

In the end this results in current z being higher, when changing tool the nozzle gets placed lower than it should. (I can upload a video to YouTube if it would help)
TOOL_BED_MESH_CALIBRATE is being used as implemented. 

SOLUTION?
- Disable bed mesh (this fixes the movement into negative direction, but isn't ideal, since having a bed mesh is nice)
- Add the following: 
[gcode_macro T0]
variable_color: ""
gcode:
  G91
  G0 Z10 F300
  G90
  BED_MESH_CLEAR
  SELECT_TOOL T=0
  BED_MESH_PROFILE LOAD=default
  G91
  G0 Z-10 F300
  G90
(I haven't tested this yet with actual printing, just with no filament and temperatures very low, and it seems to work)

Maybe there is a better option, please tell me. Is there a better way to implement this in toolchanger.py, or cfg, or is this already implemented?

[message.txt](https://github.com/user-attachments/files/19108808/message.txt)

viesturz, 2025-03-06:
The cause it not really from the mesh, it seems that restore_position is computed without accounting for the mesh.
Caused by https://github.com/viesturz/klipper-toolchanger/commit/dc4f503c8e2bcba83f0443bf6c2d4e74108abd68

0101falcon, 2025-03-06:
That would be lovely man, I mean in the end one could just do what I did and clear bed mesh between every toolchange. 

You should have seen my face when the nozzle started scrapping the build plate XD. 

Btw. new fan adjustment works great. A few other questions, have you seen this:
https://github.com/DraftShift/StealthChanger/issues/105

Is it normal that after toolchange the toolhead sometimes, stays still for 3-5 seconds, it does that for me?

Is it ok if I add some documents to tapchanger in the foreseeable future (I might also improve documentation for toolchanger and or tapchanger)?

0101falcon, 2025-03-06:
Oh and after turning on the printer for the first time, toolchange when printing fails, I get a "tool not detected" even though path hasn't been run. 

I think I can fix that by running a toolchange to all active tools in start sequence though, just curious, might be my machine though.

viesturz, 2025-03-06:
> Is it normal that after toolchange the toolhead sometimes, stays still for 3-5 seconds, it does that for me?
There is a wait for temperature in the change sequence to minimize oozing - most likely your toolhead temperature is not stable, or you have a idle temp that is lower than printing temp.

> Oh and after turning on the printer for the first time, toolchange when printing fails, I get a "tool not detected" even though path hasn't been run.
There is a current tool detection on toolchanger initialization. If you initialize on first toolchange (I do not remember how it is for that config) then that is when you would get an error.

viesturz, 2025-03-06:
> Hmmm tricky. restore_position is just current gcode_position + new tool offset, which is aimed to place nozzle at the same physical location as the previous tool. So the delta you see should be matching your tool offset (becuase gcode offset is at 0 when toolchanging) and the mesh offset is applied on top of that.
> 
> However the way the new GCODE_OFFSET is set after tool change DOES mess up the bed mesh, with potentially horrible shifts increasing with each toolchange😨 Fixing that.

This turned out to not be true, the code is still good, just convoluted. Simplified that. Doing some testing on my machine.

0101falcon, 2025-03-06:
Thanks, temperature makes sense. The initialisation not entirely, INITIALIZE_TOOLCHANGER is my 4th command in PRINT_START, after turning on skirt fans. 

Not sure what it is, maybe something on my side, either not doing bed mesh or clearing it before changing really did fix the issue, I am now able to print, so eeh, if we don't know why.

viesturz, 2025-03-06:
Tested on my machine, the Z offsets are coorect, even with mesh applied.
One issue I noticed is that manual Z offset via the UI is not added into the restore_pos, so it's restoring wrong by the manual Z offset value.

Otherwise the mesh offset is correctly applied AFAICT.

0101falcon, 2025-03-06:
Hmmm. The z offsets are also correct when I do it, as described, the "bed mesh" (on my side) does not get applied to offset itself, it is also not visible in Klipper z-value (it is an invisible adjustment on my side), it is only visible when printing the following MSG: 

  dropoff_gcode:
    {% set x = tool.params_park_x|float %}
    {% set y = tool.params_park_y|float %}
    {% set z = tool.params_park_z|float %}
    {% set fast = tool.params_fast_speed|float %}
    {% set path = tool.params_dropoff_path %}
    {% set max_z = printer.configfile.config["stepper_z"]["position_max"]|float %}
    {% set cur_z = printer.toolhead.position[2]|float %}
    RESPOND TYPE=echo MSG='Current z {cur_z}'
    RESPOND TYPE=echo MSG='Dropping off {tool.name}'

The cur_z seems to contain it for me, this offset is not visible in fine tuning Klipperscreen UI or in the z-value of Klipper. 

I will upload a video tomorrow showing this.

0101falcon, 2025-03-06:
Thanks again for taking the time.

viesturz, 2025-03-06:
> also not visible in Klipper z-value (it is an invisible adjustment on my side),
I can see it by running :

```
SET_GCODE_OFFSET Z=0.03
BED_MESH_PROFILE LOAD=default
G0 X100 Z1
GET_POSITION

// mcu: stepper_x:-5134 stepper_y:-6192 stepper_z:1498 stepper_z1:4104 stepper_z2:2170 stepper_z3:-304
// stepper: stepper_x:279.100000 stepper_y:-79.100000 stepper_z:0.990000 stepper_z1:0.990000 stepper_z2:0.990000 stepper_z3:0.990000
// kinematic: X:100.000000 Y:179.100000 Z:0.990000
// toolhead: X:100.000000 Y:179.100000 Z:0.989375 E:0.000000
// gcode: X:100.000000 Y:179.100000 Z:1.030000 E:0.000000
// gcode base: X:0.000000 Y:0.000000 Z:0.030000 E:0.000000
// gcode homing: X:0.000000 Y:0.000000 Z:0.030000
```

The toolhead.Z is the final position with bed mesh applied.

Confusingly, the gcode.z reports the gcode position + z offset 🤷 
The { printer.gcode_move.gcode_position } has the correct value however, historical reasons I suppose.

0101falcon, 2025-03-06:
Sooooo, does this mean that my issue is actually caused by what I am saying?
Because toolhead does seem to have the bed mesh applied, and all the others don't. Again if this has any effect on tool changing, well you are to say that, I don't know the code at all.

viesturz, 2025-03-06:
Pushed a change to apply the manual z adjust.

If you can paste: 
- your Z offsets for T0 & T1

Results of running this:
```
BED_MESH_CLEAR
G0 X100 Z1
GET_POSITION
BED_MESH_PROFILE LOAD=default (or something)
G0 X100 Z1
GET_POSITION
SELECT_TOOL T=1 RESTORE_AXIS=XYZ
GET_POSITION
G0 X100 Z1
GET_POSITION

```
This should give me more info on how the toolhead position works on your machine.

0101falcon, 2025-03-06:
I will do that tomorrow, I will also send that video I promised. Thanks for looking into it

0101falcon, 2025-03-07:
here console, with your commands provided above, this is before pulling yesterdays commit:

07:30
FIRMWARE_RESTART

07:30
Internal error on command:"T0"

07:30
Internal error on command:"SELECT_TOOL"

07:30
Klipper state: Shutdown

07:30
Must home axis first: 5.600 -1.000 353.000 [0.000]

07:30
Found active tool probe: tool_probe T0 ; ME: it finds it after tool not detected?

07:30
Tool not detected, expected 0. Pausing the print.

07:30
M109 run

07:30
Picking up tool T0

07:30
Activating extruder extruder

07:30
Dropping off tool T1

07:30
Current z 11.135

07:30
T0

; above this are my commands, so I switched back to T0 and got the "tool not detected" (this time it did start pickup, and tap did light up, so as far as I am concerned change was a success, the toolchange was violent however, because T1 crashed a little bit because of offset caused by what happened before... not sure why again)

07:28
mcu: stepper_x:-57198 stepper_y:24656 stepper_z:277319 stepper_z1:277319 stepper_z2:277319 stepper_z3:277319
stepper: stepper_x:68.900000 stepper_y:78.500000 stepper_z:352.970000 stepper_z1:352.970000 stepper_z2:352.970000 stepper_z3:352.970000
kinematic: X:73.700000 Y:-4.800000 Z:352.970000
toolhead: X:99.771875 Y:177.012500 Z:1.132059 E:0.000000
gcode: X:99.771875 Y:177.012500 Z:1.165000 E:0.000000
gcode base: X:-0.228125 Y:-0.287500 Z:0.165000 E:0.000000
gcode homing: X:-0.228125 Y:-0.287500 Z:0.165000

07:28
mcu: stepper_x:-57198 stepper_y:24656 stepper_z:277319 stepper_z1:277319 stepper_z2:277319 stepper_z3:277319
stepper: stepper_x:68.900000 stepper_y:78.500000 stepper_z:352.970000 stepper_z1:352.970000 stepper_z2:352.970000 stepper_z3:352.970000
kinematic: X:73.700000 Y:-4.800000 Z:352.970000
toolhead: X:99.771875 Y:177.012500 Z:1.132002 E:0.000000
gcode: X:99.771875 Y:177.012500 Z:1.164943 E:0.000000
gcode base: X:-0.228125 Y:-0.287500 Z:0.165000 E:0.000000
gcode homing: X:-0.228125 Y:-0.287500 Z:0.165000

07:28
Selected tool 1 (tool T1)

07:28
Restore_pos_z 1.165

07:28
Found active tool probe: tool_probe T1

07:28
M109 run

07:28
Picking up tool T1

07:28
Activating extruder extruder1

07:28
Dropping off tool T0

07:28
Current z 0.9670585259609491

07:28
mcu: stepper_x:-23854 stepper_y:-272 stepper_z:2943 stepper_z1:2943 stepper_z2:2943 stepper_z3:2943
stepper: stepper_x:277.300000 stepper_y:-77.300000 stepper_z:10.000000 stepper_z1:10.000000 stepper_z2:10.000000 stepper_z3:10.000000
kinematic: X:100.000000 Y:177.300000 Z:10.000000
toolhead: X:100.000000 Y:177.300000 Z:0.967059 E:0.000000
gcode: X:100.000000 Y:177.300000 Z:1.000000 E:0.000000
gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
gcode homing: X:0.000000 Y:0.000000 Z:0.000000

07:28
mcu: stepper_x:-23854 stepper_y:-272 stepper_z:2943 stepper_z1:2943 stepper_z2:2943 stepper_z3:2943
stepper: stepper_x:277.300000 stepper_y:-77.300000 stepper_z:10.000000 stepper_z1:10.000000 stepper_z2:10.000000 stepper_z3:10.000000
kinematic: X:100.000000 Y:177.300000 Z:10.000000
toolhead: X:100.000000 Y:177.300000 Z:1.000000 E:0.000000
gcode: X:100.000000 Y:177.300000 Z:1.000000 E:0.000000
gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
gcode homing: X:0.000000 Y:0.000000 Z:0.000000

07:28
BED_MESH_CLEAR
G0 X100 Z1
GET_POSITION
BED_MESH_PROFILE LOAD=default
G0 X100 Z1
GET_POSITION
SELECT_TOOL T=1 RESTORE_AXIS=XYZ
GET_POSITION
G0 X100 Z1
GET_POSITION

T0:
gcode_x_offset: 0
gcode_y_offset: 0
gcode_z_offset: 0

z_offset: -1.25 #-0.957 ; I use this to adjust for thermal expansion, I will add another ticket discussing this issue, I have some ideas!

T1:
gcode_x_offset: -0.228125 # -0.223000
gcode_y_offset: -0.287500 # -0.265625
gcode_z_offset: 0.165000 # 0.132500

z_offset: -1.045

Here is the link, I will later today SSH into the printer, update toolchanger and retry it again. 

Here is my issue in video format
https://www.youtube.com/watch?v=M7GIgOhSjNE

0101falcon, 2025-03-07:
Oh yeah, and I say that my nozzle is being set higher at the start of the print (more than 0.2mm)

I do that by adjusting this:
z_offset: -1.25 #-0.957 

to this:
z_offset: -1.5 

And then homing with T0, one could also do a G92 after homing, in the end I just need it to be higher so I can run a couple of tool changes.

0101falcon, 2025-03-10:
> BED_MESH_CLEAR
> G0 X100 Z1
> GET_POSITION
> BED_MESH_PROFILE LOAD=default (or something)
> G0 X100 Z1
> GET_POSITION
> SELECT_TOOL T=1 RESTORE_AXIS=XYZ
> GET_POSITION
> G0 X100 Z1
> GET_POSITION

I ran this with the new push, I got this:

15:44
mcu: stepper_x:-29770 stepper_y:5492 stepper_z:-14289 stepper_z1:-14972 stepper_z2:-15690 stepper_z3:-14057
stepper: stepper_x:238.725000 stepper_y:-39.675000 stepper_z:1.142937 stepper_z1:1.142937 stepper_z2:1.142937 stepper_z3:1.142937
kinematic: X:99.525000 Y:139.200000 Z:1.142937
toolhead: X:99.796875 Y:24.668750 Z:1.138183 E:0.000000
gcode: X:99.796875 Y:24.668750 Z:1.176250 E:0.000000
gcode base: X:-0.203125 Y:-0.331250 Z:0.176250 E:0.000000
gcode homing: X:-0.203125 Y:-0.331250 Z:0.176250
15:44
mcu: stepper_x:-29770 stepper_y:5492 stepper_z:-14289 stepper_z1:-14972 stepper_z2:-15690 stepper_z3:-14057
stepper: stepper_x:238.725000 stepper_y:-39.675000 stepper_z:1.142937 stepper_z1:1.142937 stepper_z2:1.142937 stepper_z3:1.142937
kinematic: X:99.525000 Y:139.200000 Z:1.142937
toolhead: X:99.796875 Y:24.668750 Z:1.138257 E:0.000000
gcode: X:99.796875 Y:24.668750 Z:1.176324 E:0.000000
gcode base: X:-0.203125 Y:-0.331250 Z:0.176250 E:0.000000
gcode homing: X:-0.203125 Y:-0.331250 Z:0.176250
15:44
Selected tool 1 (tool T1)
15:44
Restore_pos_z 1.17625
15:44
Found active tool probe: tool_probe T1
15:44
Verifying Tool Now ; this is my comment which I added since I get errors from time to time when changing tools the manual changing didn't fix it, turns out...
15:44
M109 run
15:44
Picking up tool T1
15:44
Activating extruder extruder1
15:44
Dropping off tool T0
15:44
Current z 0.9619326241134751
15:44
mcu: stepper_x:-15966 stepper_y:55840 stepper_z:-7197 stepper_z1:-7880 stepper_z2:-8598 stepper_z3:-6965
stepper: stepper_x:325.000000 stepper_y:275.000000 stepper_z:10.007937 stepper_z1:10.007937 stepper_z2:10.007937 stepper_z3:10.007937
kinematic: X:300.000000 Y:25.000000 Z:10.007937
toolhead: X:100.000000 Y:25.000000 Z:0.961933 E:0.000000
gcode: X:100.000000 Y:25.000000 Z:1.000000 E:0.000000
gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
gcode homing: X:0.000000 Y:0.000000 Z:0.000000
15:44
mcu: stepper_x:-15966 stepper_y:55840 stepper_z:-7197 stepper_z1:-7880 stepper_z2:-8598 stepper_z3:-6965
stepper: stepper_x:325.000000 stepper_y:275.000000 stepper_z:10.007937 stepper_z1:10.007937 stepper_z2:10.007937 stepper_z3:10.007937
kinematic: X:300.000000 Y:25.000000 Z:10.007937
toolhead: X:100.000000 Y:25.000000 Z:1.000000 E:0.000000
gcode: X:100.000000 Y:25.000000 Z:1.000000 E:0.000000
gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
gcode homing: X:0.000000 Y:0.000000 Z:0.000000

One more thing, this has partially something to do with #52, when I add a "SET_GCODE_OFFSET Z_ADJUST=+0.25"
To my machine start Gcode, it does adjust the layer, but the following happens:

First layer at 0.20
Changing Tool
Layer drops to -0.05
Next Layer at 0.4
Changing Tool 
Layer drops to 0.15
Next Layer at 0.6

This happens through the entire print, is this expected behaviour???

0101falcon, 2025-05-29:
Hey, have you had any time to check this?
Haven't heard any updates in a while...

Contomo, 2025-06-11:
https://github.com/viesturz/klipper-toolchanger/issues/69#issue-2973654544