viesturz/klipper-toolchanger #49: Unknown config object 'fan_generic multi_fan T0_partfan'

leezus-nz, 2025-03-06:
I've updated today and unfortunately have this error now:
Unknown config object 'fan_generic multi_fan T0_partfan'

I've tried to track down the issue and compare against the example config – doesn't seem to help.

0101falcon, 2025-03-07:
Replace mutlifan with this under toolhead_n.cfg

#[multi_fan T0_partfan]
#pin: et0:PA1
[fan_generic FAN0]
pin: et0:PA1

replace fan: multifan T0_partfan with this:
[tool T0]
tool_number: 0
extruder: extruder
params_type: 'rods_mini_hookon'
#fan: multi_fan T0_partfan
fan: FAN0
gcode_x_offset: 0
gcode_y_offset: 0
gcode_z_offset: 0
Position with the toolhead aligned with the dock
params_park_x: 5.6
params_park_y: -5.0
params_park_z: 353.0

I found that this worked for me. Multifan has been replaced, see latest commit.

viesturz, 2025-03-07:
> Not sure if that's the "correct" solution, but it works

Example config update coming up

leezus-nz, 2025-03-07:
> Replace mutlifan with this under toolhead_n.cfg
>
> #[multi_fan T0_partfan] #pin: et0:PA1 [fan_generic FAN0] pin: et0:PA1
>
> replace fan: multifan T0_partfan with this: [tool T0] tool_number: 0 extruder: [...]

Seems to be working, thanks so much!

viesturz, 2025-03-09:
Updated the example configs as well, Thanks for reporting!

closed 2025-03-09 by viesturz