viesturz/klipper-toolchanger #53: Documentation for transfer_fan_speed wrong?

piposa, 2025-03-07:
First of all I like the new implementation.  I also like that I can still define my own M106 gcode_macro which overwrites the internal (tested, ok). I assume that the transfer_fan_speed yes/no is independent of the M106 internal/external implementation (would be good, not tested yet).

There is a minor documentation issue:
"transfer_fan_speed: True" is listed under the [tool] section, not [toolchanger]. I put it under [toolchanger] and only tested this, but I would not understand what happens if I have it only under one tool (transferred from or to this tool or both or only if both tools have it set to true, ...). I assume this will anyway not work, because it has to be under [toolchanger]? If yes documentation should be changed, if yes behavior should be explained.

Thanks again for all the quick fixes!

0101falcon, 2025-03-08:
Yeah pretty sure you are right on this one, consider:
https://github.com/viesturz/klipper-toolchanger/commit/e2565b5706b537ab42a5d142a7f6ae71b4f34b5d

We see that:
self.transfer_fan_speed = toolchanger.transfer_fan_speed

Which calls it from toolchanger, don't understand the syntax yet but yes. 

Add:
transfer_fan_speed: True
To toolchanger.cfg

viesturz, 2025-03-09:
Whoopsie, you are right, it's for the toolchanger section.

commit aa3af87d (2025-03-09): Fix transfer fan speed documentation. Fixes #53
closed 2025-03-09 by viesturz