viesturz/klipper-toolchanger #54: Crash when homing with updated tool_probe_endstop feature

noobydp, 2025-03-08:
https://github.com/viesturz/klipper-toolchanger/blob/8de49b70fdb7b6832ff08370a74b2e80def8898c/klipper/extras/tool_probe_endstop.py#L251C5-L251C31
This line is causing klipper to crash when homing with an error: 
> TypeError: ProbeRouter.on_error() got an unexpected keyword argument 'triggered'

changing to this avoids the crash, but I'm not entirely sure why the error is happening to start with.: 
`def on_error(self, *args, **kwargs)`

This was also mentioned in a comment on #35

viesturz, 2025-03-09:
The core issue is that you are trying to home before toolchanger is initialized and the endstop does not know which probe to use.
The on_error method was supposed to print that but had a bug fixed that.

closed 2025-03-09 by viesturz

noobydp, 2025-03-10:
Thanks.

I did a bunch of tests after I changed that code and realised what youre saying, but I made sure the tool was initialized and I still got endstop still triggered errors.
I'll redo the tests and open a new ticket with all my config and logs when I get a chance

commit 504f58c7 (2025-03-09): tool_probe_endstop.py fix on_error not accepting kwargs. fixes #54