viesturz/klipper-toolchanger #55: Problem with new install?!?!?

printicus, 2025-03-10:
Hi

I reinstalled Klipper and all on my 2.4, and i cannot get tools to change... Klipper goes in panic mode. Looked trough the files and in Toolchanger.cfg allota code is in brown color. Is that how its supposed to be?

[image]
[image]
[image]
[image]

Python version 3.11 is on my printer

viesturz, 2025-03-10:
Let's start with following line 11

printicus, 2025-03-10:
Does not matter... the pictures are from a new paste of the entire code. With or without the tool path the color remains the same... If you see in the first picture the brown text is allso above the lines for tool path lines. Im not home right now, but I will try again tonight. Maybe i fucked up somehow....

printicus, 2025-03-11:
Somehow I did make it work. This is the "toolchanger.cfg" that works for me.

[toolchanger.cfg.txt](https://github.com/user-attachments/files/19182021/toolchanger.cfg.txt)
I deleted some spaces that where in front of code so it would change to the proper color.
The big modification is that i needet to change the "before_change_gcode:" and "after_change_gcode:" to "beforechange_gcode:" and "afterchange_gcode:". That underline was not liket by my printer... :D. And i needet allso to make that change in two of the .py files that are in the "extras" folder in the printer. But now it works.

Thank You

0101falcon, 2025-03-14:
so where did you remove spaces?

I have the same color issue, but my code actually works like that, so I am unsure if it is affecting functionality...

printicus, 2025-03-20:
My bad... im a moron... turns out it was a bad install and/or a bad sdcard.   After a while the printer started going crazy, slow responds, stuttering moves, long time to start the printer. I tryed to reinstall but the sd card died. New sd card, new install and all goes pefect from the begining. Nothing wrong with the code. 
Your kung-fu is the best.

Contomo, 2025-04-04:
may this issue be closed then?

closed 2025-04-09 by printicus