viesturz/klipper-toolchanger #57: sporadic exception during tool changes, error in rounded path curve calculation!

idonasch, 2025-03-12:
Under certain conditions the rounded path calculation misses the tangential endpoint of the arc and tries to move the head outside of the build volume causing the exception. This is rare but fatal condition and can happen when the too change position happens relative to certain position defined  in the rounded_G0 path. The klipper logs were not helpfull because of the many look-a-head moved roundedG0 produces and took me month to home in on them.
My solution to fix the problem was to replace the tangential arc method with a Bezier curve. I added a python simulator to plot the original and Bezier function on a move that produced the exception.

[image]

[rp.zip](https://github.com/user-attachments/files/19207381/rp.zip)

the round_path.py is a mangled debug version of the extras/rounded_path.py of the repository and has import sim and besier added as well as debug stuff to plot both curves. in the production code  the 'def bezier()' function body just replaces the 'if use_bezier & xxx' blocks inline.

I need the to be validated by the original owner of the code and my implementation therefore should only be a hint.

remark: using an arc to round the segment corner will not work if the tangential points are at different distances from the corner.

viesturz, 2025-03-19:
Thanks for the image, it looks like the curve was correct, but the angle was way too much (looks like about +180 degrees too much).

Hi, if you can share the stack trace of the exception (should be in your klippy.log).

Bezier curves have a problem of overshooting, which plain arcs do not.

idonasch, 2025-03-20:
Here is the Trace of one of the crashes, Only the first stack trace is of interest, the following are the consequence of moving outside of the build area resulting in the 'Must home axis first'. I already added some more descriptive debug output but the error happened way further down in the future path already pushed to the MCU and that gave me almost no indication of what actually happened until I ran the simulation and saw the arc overshooting into the negative-X.

[T0.zip](https://github.com/user-attachments/files/19360132/T0.zip)

Activating extruder extruder
Must home axis first 1:[1.000000 > -1.000000]: 348.400 57.929 284.800 [0.000]
Internal error on command:"SELECT_TOOL"
Traceback (most recent call last):
  File "/home/ingo/klipper/klippy/extras/toolchanger.py", line 470, in run_gcode
    template.run_gcode_from_command(context)
  File "/home/ingo/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/ingo/klipper/klippy/gcode.py", line 227, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/ingo/klipper/klippy/gcode.py", line 212, in _process_commands
    handler(gcmd)
  File "/home/ingo/klipper/klippy/gcode.py", line 140, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/ingo/klipper/klippy/extras/rounded_path.py", line 123, in cmd_ROUNDED_G0
    self._lineto(ControlPoint(x = gcmd.get_float("X", currentPos[0]),
  File "/home/ingo/klipper/klippy/extras/rounded_path.py", line 137, in _lineto
    self._flush_buffer(len(self.buffer) -2)
  File "/home/ingo/klipper/klippy/extras/rounded_path.py", line 178, in _flush_buffer
    self._arc(self.buffer[i+1], self.buffer[i],self.buffer[i+2])
  File "/home/ingo/klipper/klippy/extras/rounded_path.py", line 235, in _arc
    self._g0p(c, _vadd(center, rotspoke))
  File "/home/ingo/klipper/klippy/extras/rounded_path.py", line 252, in _g0p
    self.real_G0(self.G0_cmd)
  File "/home/ingo/klipper/klippy/extras/gcode_move.py", line 143, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/home/ingo/klipper/klippy/extras/bed_mesh.py", line 217, in move
    self.toolhead.move([x, y, z + self.fade_target, e], speed)
  File "/home/ingo/klipper/klippy/toolhead.py", line 477, in move
    self.kin.check_move(move)
  File "/home/ingo/klipper/klippy/kinematics/corexy.py", line 78, in check_move
    self._check_endstops(move)
  File "/home/ingo/klipper/klippy/kinematics/corexy.py", line 71, in _check_endstops
    raise move.move_error("Must home axis first %d:[%f > %f]"%(i, self.limits[i][0],self.limits[i][1]))
gcode.CommandError: Must home axis first 1:[1.000000 > -1.000000]: 348.400 57.929 284.800 [0.000]

idonasch, 2025-03-20:
...
> Bezier curves have a problem of overshooting, which plain arcs do not.
not that I'm aware off.

a Bezier curve might not be the final answer, but it solves the problem of tangential entry and exits and does not need any trigonometric functions that are known to have edge- and resolution issue, e.g. tan(). For now I will stick with Bezier curves but I'm open for anything that works.

BTW, I tried to use arcs with just linear algebra but shoot my self in the foot when I tried to use arc entry- and exit point at different distances from the corner (different center points!!), maybe an ellipse would solve the problem with dual arc centers. If required I could elaborate more on this topic maybe on a different channel.

Contomo, 2025-04-10:
+1 from me. 
ROUNDED_G0 X=184.79999999999998 Y=179.6 Z=9.0 F=4000.0 D=0
ROUNDED_G0 X=184.79999999999998 Y=179.6 Z=14.0 F=20000.0 D=5
ROUNDED_G0 X=184.79999999999998 Y=179.6 Z=159.15 F=20000.0 D=200
ROUNDED_G0 Y=210.0 Z=309.3 F=20000.0 D=90
ROUNDED_G0 X=1.0 Y=105.0 Z=309.3 F=20000.0 D=30
is another crashing sequence :P

Contomo, 2025-07-13:
[image]

```
    {'X': 312.0, 'Y':  0.0, 'Z': 5.5, 'F': 12000.0, 'D': 0},
    {'X': 320.0, 'Y': -4.0, 'Z': 5.5, 'F': 12000.0, 'D': 0},
    {'X': 328.0, 'Y':  0.0, 'Z': 6.0, 'F': 12000.0, 'D': 0},
    {'X': 336.0, 'Y': -4.0, 'Z': 5.5, 'F': 12000.0, 'D': 0},
    {'X': 344.0, 'Y':  0.0, 'Z': 5.5, 'F': 12000.0, 'D': 0},
    {'X': 336.0, 'Y':  4.0, 'Z': 5.5, 'F': 12000.0, 'D': 0},
    {'X': 328.0, 'Y':  0.0, 'Z': 6.0, 'F': 12000.0, 'D': 0},
    {'X': 320.0, 'Y':  4.0, 'Z': 5.5, 'F': 12000.0, 'D': 0},
    {'X': 312.0, 'Y':  0.0, 'Z': 5.5, 'F': 12000.0, 'D': 0},
```
first coord is starting pos, last is always D=0

i dont know if this helps with anything... i just think its visually helpful to see.