viesturz/klipper-toolchanger #59: No tool detected

TheEvilTweety, 2025-03-14:
hi 
for a few days now, i have always had the problem that when it changes, it says no tool detected. but if i change manually and then execute VERIFY_TOOL_DETECTED, there are no problems. 
as far as i know i have tried everything. updates and firmware are all up to date and i have also updated the toochanger.

0101falcon, 2025-03-15:
I have a similar issue, my idea of using a M400 didn't solve the issue:

I have an issue when I changed my tools the first time, I would get a Tool not found, after restarting Klipper and trying again it would work. 

I decided to add a Output and realized that Klipper was preloading the VERIFY_TOOL_DETECTED, it should actually do it in the step where it says verify:1 in the pickup code. Yet for me it verified before the pickup path was run. 

My idea was to remove the verify from the pickup path, and place it below the loop that runs the path.
I also added a M400 above the VERIFY_TOOL_DETECTED (wait until everything before has been run) 

My changed code: (I have no more verify in pickup, anyone has an idea why tool not detected happens before Verifying tool???)
{% for pos in path %}
      {% set speed = tool.params_path_speed|float * (pos.get('f', 1.0)|float) %}
      G0 {% if 'x' in pos %}X{x + pos['x']|float}{% endif %} {% if 'y' in pos %}Y{y + pos['y']|float}{% endif %} {% if 'z' in pos %}Z{z + pos['z']|float }{% endif %} F{speed}
      {% if 'verify' in pos %}
        RESPOND TYPE=echo MSG='Verifying Tool Now'
        _VERIFY_TOOL_DETECTED T={tool.tool_number}
      {% endif %}
    {% endfor %}
    M400
    RESPOND TYPE=echo MSG='Verifying Tool Now2'
    _VERIFY_TOOL_DETECTED T={tool.tool_number}

This is my Console:

Internal error on command:"T0"
Internal error on command:"SELECT_TOOL"
Klipper state: Shutdown
Must home axis first: 5.600 101.716 353.000 [55.362]
Found active tool probe: tool_probe T0
Verifying Tool Now2
Tool not detected, expected 0. Pausing the print.
M109 run
Picking up tool T0
Activating extruder extruder
Dropping off tool T1
Current z 20.43671299999999
Selected tool 1 (tool T1)
Restore_pos_z 20.18671299999999
Found active tool probe: tool_probe T1
Verifying Tool Now2
M109 run
Picking up tool T1 

Do you need a log to realise what's wrong?

viesturz, 2025-03-19:
Hmm, VERIFY_TOOL_DETECTED waits for any pending moves to finish before verifying.
I see that this is also a problem in other places, the Klipper code base is sprinkled with stuff like 
```
        toolhead.wait_moves()
        toolhead.dwell(0.100)
        toolhead.wait_moves()
```

Adding same for VERIFY_TOOL_DETECTED

closed 2025-03-19 by viesturz