viesturz/klipper-toolchanger #60: Kalico support

dderg, 2025-03-18:
I would like to switch from Klipper to Kalico, and they made clear that they don't plan to bring probe class changes from klipper in. Would it be possible to make the library support both Klipper and Kalico?

viesturz, 2025-03-19:
Let me know how it goes. Do you know if Kalico has any plans to merge in the Klipper probing code refactor?

dderg, 2025-03-20:
they said they have plans to never merge it

dderg, 2025-04-08:
So far it is working fine. I will keep merging your changes from time to time to keep parity

rogerlz, 2026-01-14:
> Let me know how it goes. Do you know if Kalico has any plans to merge in the Klipper probing code refactor?

For now, no.. but would you both be willing to work together and make the code compatible with both Kalico and Klipper, similar to other probe modules such as beacon/eddy-ng?