viesturz/klipper-toolchanger #61: Tool position after toolchange

printicus, 2025-03-22:
I did not know how to properly name this title, so I hope I cand explain it better here...
During a print, after a tool change, the "new" tool comes to the "old" tool z level, but not x and y pozition (if I understood correctly). The x and the y position is no problem, but the z would be better if it would stop some 10-50mm higher, and from there to go where it needs to start printing. How it is now, if a print is in that place where the tool commes down, the nozzle would park o the print and from there would go dragging the nozzle and start the print. Or, maybe , would be better if the tool would go, on the way down, direct to the coordonates where it needs to start printing.

Contomo, 2025-03-22:
`t_command_restore_axis:`
yours likely is `t_command_restore_axis:XYZ`
depending on what slicer you are using you can get issues with not restoring that (most dont have issues with X and Y, as there is an immediate call for G0 X Y after toolchange anyway, but orca has issues with Z for example)
(ie if you dont restore Z, it starts printing mid air, till the next G0 Z is called)

i recommend setting it to either Z or nothing, depending on what your slicer supports.

printicus, 2025-03-22:
I use both SS and PS. I tryed Orca but cannot stand the layout....

It Is "t_command_restore_axis: Z"
The z height happens if I print or not. Just by changing tools whitout printing the "new" z will be the z from where the previous tool started the tool changing routine(I dont care about x and y). This is what i want to say: the new tool should stop at a higher position after a toolchange, not at the same z hight as the previous tool because this happens:

[image]

Contomo, 2025-03-22:
just replace in your `pickup_gcode:`
```
    ROUNDED_G0 Y={tool.params_safe_y} F={fast} D=20
    {% if 'Z' in restore_position %}
      ROUNDED_G0 Z={restore_position.Z} F={fast} D=150
    {% endif %}
    {% if 'X' in restore_position %}
      ROUNDED_G0 X={restore_position.X} F={fast} D=1000
    {% endif %}
    {% if 'Y' in restore_position %}
      ROUNDED_G0 Y={restore_position.Y} F={fast} D=0
    {% endif %}
    ROUNDED_G0 D=0
```
`ROUNDED_G0 Z={restore_position.Z} F={fast} D=150` => with `ROUNDED_G0 Z={restore_position.Z + **10**} F={fast} D=150`

printicus, 2025-03-23:
I will try this. Thank you, Sir.

printicus, 2025-03-24:
Ok.. it kinda works but does not.... the +10 in that z restore position is only shows in the movement like a short hickup and still goes on to restored z position.

Contomo, 2025-03-24:
sounds like thats a slicer thing. check your gcode file and search for the "TX" command of your tool, and check what gcode comes right after that, if its the same position as the one where it stopped. it is your slicer.

printicus, 2025-03-24:
Nothing to do with slicer. Just by changing tools in mainsail.

Contomo, 2025-03-25:
would it be possible for you to share anything related to your code? its kinda hard guessing.

printicus, 2025-03-25:
I dont know what code to share.... I have done no modification to original toolchanger code, other than dropoff and pickup paths...

Contomo, 2025-03-25:
> I have done no modification to original toolchanger code, other than dropoff and pickup paths...

and the z+10 thingy right?
the video is of what? a TX command? or test tool docking or?

post your current toolchanger.cfg if you can.

idk it may be related to the internals saving gcode state.
a kinda hacky way to fix it would be to move z+10 after the tx call?
because the internal one restores the gcode state (old position) but doesnt move.
something must be invoking a move for you i guess?
calling z+10 right after tx would tell it to move to gcode state old +10 (which would be where you are now.
and only then finishing going on.
thus it shouldnt move back the original z height.

but i really dont know whats going on in that video. its like yeah... i believe you that its doing that. the video is just like you saying that again.
it doesnt really help to track down the issue.

printicus, 2025-03-25:
Yeah, the modification I made are the dropoff and pickup paths, and the +10 (in the video is +50). Im at work right now, when i get home i will post the toolchanger.cfg

printicus, 2025-03-25:
Looking now in toolchanger.cfg I see I have another modification. I had the same error like here:
https://github.com/viesturz/klipper-toolchanger/issues/59
and I applied the solution from there. Tryed to revert to original code and had this result:

https://github.com/user-attachments/assets/54bfa139-2170-45ce-8f63-1f825d965e22

[klippy.log](https://github.com/user-attachments/files/19458057/klippy.log)

But, the toolhead stopped at +50 whit this error... 

So my toolchanger.cfg is like this now:

[Toolchanger.cfg.txt](https://github.com/user-attachments/files/19458076/Toolchanger.cfg.txt)

Looks like this modification breaks the move i want...

Contomo, 2025-03-26:
is "verify tool detected" actually valid to be used with tool probe endstops?
START_TOOL_PROBE_CRASH_DETECTION T={tool.tool_number} i thought was the way to enable crash detection again.
from what i got verify is only used for tools that only use "detected" and not the probe endstop?

printicus, 2025-03-26:
Thats over my pay grade....(I mean i dont know, i jyst start lerning...) The original code is comented out. the new code just addet: M400
    RESPOND TYPE=echo MSG='Verifying Tool Now2'
    _VERIFY_TOOL_DETECTED T={tool.tool_number}

what seems to do the trick whit detecting the tool. I use detection pin, not z endstop,  for tool detection.

printicus, 2025-03-26:
Yeah... I just tryed one more time with the original "{% for pos in path %}" code and now it works... seems its a restart problem. I will see next time I start the printer.

Contomo, 2025-03-26:
https://github.com/viesturz/klipper-toolchanger/blob/6c625dfb6029f3adf8242a963f659bd05c8d77a4/klipper/extras/toolchanger.py#L322

currently restoring the axis internally even tho we provide examples of restoring it via macro.

https://github.com/viesturz/klipper-toolchanger/blob/6c625dfb6029f3adf8242a963f659bd05c8d77a4/examples/per%20tool%20probe/toolchanger.cfg#L89-L100

printicus, 2025-03-26:
My setup is probe on all toolheads.
Your solution with "G91 -> G0 Z5 -> G90" works like a charm, in swiching tools in mainsail and while printing. Thank You.

Now, because I see you put efort in it... can I ask you for more?   :)

Contomo, 2025-03-27:
depends on your definition for more

printicus, 2025-03-27:
:)

I use my own design for shuttle and backplate(inspired on stealtchanger). the diference is that I use pogo pins for electrical contact between shuttle and backplate. In this mode i can have x and z probes on the shuttle (and maybe other thing too). My problem, for now, is that i cannot use properly the x endstop... In my dum head I tought that if i put 

[stepper_x]
endstop_pin: nhk0:gpio13
position_min: 0

in each toolhead config, each whit his own pin, when the toolhead is set active then also the x-enstop is set to that toolhead... but no.
So.. my question for you: is it possible for when a toolhead is set active, beside all other thing that are restored(z-offset, imput shaper, pid.. etc) the x-enstop pin to be also set for that tool?

For reference here are my designs. They are old, my new design is not out yet.
https://www.thingiverse.com/thing:6731214

Contomo, 2025-03-28:
honestly no clue, maybe virtual x endstop?
biggest problem ima is that you would also have to calibrate all the x endstops and tell klipper the new true X0 positionb (similar t0 how its done with Z)
easiest option imo is just to use sensorless

printicus, 2025-03-28:
No need for calibration. It would be one microswich on the shuttle to which each toolboard will connect when they're tool is on the shuttle.

Contomo, 2025-04-05:
may this be closed then?