viesturz/klipper-toolchanger #62: Multiple tools detected: ['tool T0', 'tool T2', 'tool T3', 'tool T4', 'tool T5']

colin342, 2025-03-22:
I used tapchanger succesfully in the past no issues. But since the last update made it possible to use different probes, i wanted to give this a try.
So I updated to the new firmware version: Klipper-toolchanger v0.0.0-111-g00daf38a-inferred.
And changed all the toolhaed file so that only Tool 0 has a tool.

All tool probe sections have been removed.
and detection pin has been added to the tool_tX sections

But when I homed for the first time i got the error that tool 0 is expected but tool 1 is seated, this while tool 0 defentily was sitting in the carraige.
When enabling all the other tools I got the error that multiple tool are detected.

I currently am a bit stumbled on what is going on here

[image]
[image]
[image]
[image]
[image]
[image]

Contomo, 2025-03-22:
im pretty sure that if you check your previous config youll see your old definitions eithers as ^et5:PB6 or as !et5:PB6
you may be missing an inversion or pullup on your detect pins (regardless of what your goal is)

colin342, 2025-03-22:
The goal is to have a working toolchanger where the used probe is a Cartogrpaher.
I currently have one installed on toolhead 0 alone. so I switched over from the old (per tool probe) to the new configuration of tapchanger like in the examples: probe on T0.

I tried to use a ! or ^ or ^! but I always get the same issue when I want to home it is throwing the multiple tool detected error

When I  use the configuration per tool probe I don't have any issue at all

printicus, 2025-03-22:
Ok, but what is this detection pin you are using? A swich or something that is trigered by the specific tool that is om the carriage?

viesturz, 2025-03-23:
Can you post your config please, Thanks!

colin342, 2025-03-23:
I am using an EBB36 v1.2 and I am using pin PB6. The detection sensor are optical sensor same as used in the tapchanger github rep

Underneath you can find the config files

[printer.txt](https://github.com/user-attachments/files/19410015/printer.txt)
[ET5.txt](https://github.com/user-attachments/files/19410016/ET5.txt)
[ET4.txt](https://github.com/user-attachments/files/19410013/ET4.txt)
[ET2.txt](https://github.com/user-attachments/files/19410018/ET2.txt)
[ET3.txt](https://github.com/user-attachments/files/19410012/ET3.txt)
[ET1.txt](https://github.com/user-attachments/files/19410014/ET1.txt)
[ET0.txt](https://github.com/user-attachments/files/19410017/ET0.txt)

viesturz, 2025-04-12:
Have you tried to invert the pin? The example configurations are all with the pin high when tool not mounted.
Most likely your sensor is pin low when not mounted.

colin342, 2025-04-13:
I Inverted the pins as you suggested and also updated to the latest version and the problem of detecting multiple toolheads is gone now.
But now I run in to the following problem:

[image]

I used to the toolchanger.cfg from examples/dock location/fixed
Here I see that _VERIFY_TOOL_DETECTED is called, but the instructions say that when only a probe is installed on T0, tool_detection.cfg should be left out.

When looking at the homing.cfg I also noticed that here it is calling VERIFY_TOOL_DETECTED (without _ in front)
So I removed the _ from the toolchanger.cfg en this solved the above issue.

Is this something that needs to be updated on the github page or did I take the wrong toolchanger.cfg file (docks are in my case in a fixed top location)