viesturz/klipper-toolchanger #63: Feature Request/maybe exists?: runtime gocde offset modification

Contomo, 2025-03-22:
long story short. klipper has a way to set GCODE offsets. yet the toolchanger offers no way to change those. it handles it in a closed internal system.
if you call set gcode offset for a tool, to set its gcode offset (for example if you use a save variables file to store offsets, or want to tweak the offset during runtime) toolchanger will always revert to those original gcode offsets defined in the tool. (which can be an issue during toolchange position restore as that uses those old offsets)

my idea: either move to params to allow editing by setting gcode params. or add a command to offset, or simply overwrite the values toolchanger is using/grabbing them currtently from.

Contomo, 2025-03-23:
Something id like to know before i put time into something which will later not be accepted anyway:
currently half is handled internally by the plugin, and half isnt.  
as an example: [the extra context](https://github.com/viesturz/klipper-toolchanger/blob/e37603bfe7f355c8b825f3fe7d3733c754f74668/klipper/extras/toolchanger.py#L293-L300) is handling all gocde offsets already that will be going on internally, yet something like [input shaper](https://github.com/viesturz/klipper-toolchanger/blob/e37603bfe7f355c8b825f3fe7d3733c754f74668/examples/per%20tool%20probe/toolchanger.cfg#L37) is handled in the config by the user.
Why is it that some things are split, and some arent?
having the restore position calculations going on in the background, but the actual restoration of the position going on in your macro?
[the gcode offset in homing](https://github.com/viesturz/klipper-toolchanger/blob/e37603bfe7f355c8b825f3fe7d3733c754f74668/examples/per%20tool%20probe/homing.cfg#L42) is handled inside of a macro same goes for the [probing z offset](https://github.com/viesturz/klipper-toolchanger/blob/e37603bfe7f355c8b825f3fe7d3733c754f74668/examples/per%20tool%20probe/homing.cfg#L40) while all of the initialization is done sorta inside.

i just have a hard time figuring out what is supposed to be built in in accessible to the users, and what is supposed to be handled outside by the user.

viesturz, 2025-03-23:
Hi, thanks for all the writeup. It definately makes sense to move repetitive things out of the config/macros and into the core.

**directive from you as to what this is supposed to be structured**

My goal was and still is to keep the python code minimal, low-ish level and have significant work done by the macros.
The bar for core features is ~
 
 *  is there a clear right way to do this
 * is this something most toolchangers need
 * or is this called out by enough people 

I was expecting people to develop their own macros for various configurations, which seems to not be the case.

This continues to evolve based on feedback, so curious to hear how do you want this structured.

**Documentation** - happy to accept pull requests that update the documentation.

**Gcode offset saving** - techncically this is trivial to implement and I would prefer to just save the offset values of previous tool during toolchange. However Z offset is also associated with bed probe microstepping, which applies to all tools equally and hence it's not clear what the "correct"  behavior should be.

**Tool verification** 

Aagain a bit of a gray zone, if anyting I would like to hear your preference. The toolchanger.verify_tool_pickup offers a basic way to ensure the printer does not destroy itself. VERIFY_TOOL_DETECTED is a generic utility to perform all kinds of checks, during toolchange or at startup.

Contomo, 2025-03-23:
> I was expecting people to develop their own macros for various configurations, which seems to not be the case.
is a tricky one hehe. that is the case, its just that theres no real way to do it. 
so for example something i just recently fully switched over to is a [macro implementation](https://github.com/joseph-greiner/klipper_tc_automatic_offset_calibration) of the outputs that the tool calibration offers `(tools_calibrate.last_?_result)`
the problem with this already is. that there is _(why this github issue exists)_ that there is no actual ways to tell the toolchanger our offsets. so a simple move of offset handling into a macro, results in a complete collapse of the internal calculations made for the `restore_position`
which wouldnt be an issue if we just did it in the pickup_gcode ourselfs, however after that it gets zerod again, from both the restore state, and the toolchanger trying to apply the offsets. thus needing to later set it once again. which suddenly not only moved out all the internals, but has also start to work actively against them, thus fully detatching from a normal config anyone can just adapt or add to theirs. (the "all homing" is actually fully implementable with user provided values for z offset and gcode z offset already, since its all just macros)
i dont think the gccode offset saving as it is right now is bad. it offers users a good entrance which works. 
i also havent experienced or heard any issues with the global Z offset adjustment. in fact. this is how its handled in the [mentioned impementation](https://github.com/joseph-greiner/klipper_tc_automatic_offset_calibration/blob/f93547803c870640367c8fcd88f9d9e7cdd03b32/tool_file_modifications.cfg#L38-L46) aswell.
so imo, do most toolchanges need to calibrate tools during runtime? probably not. Will some want it anyway? most certainly.
so my idea on how to tackle this the best way:
- we keep on handling the neccecities that every FDM toolchanger will need internally
- add possibilities for people that want to extend on that to change internals without actually changing internals.

so right now what ive seen in more advanced implementations either makes use of fully decoupling the gcode sections and doing their own (ie draft shift) or only using small parts of it (as the correct implementation of the offset saving in a save file does)
my current config is... a complete unhinged mess of macros calling macros.

so as to the tool verificiation part. i dont mind it, its just there, exactly that, its optional.
however to new users it may seem as the "detected tool" is in fact that, the tool detection. 
the awnser to the question, is my toolhead on the shutte?
so for that? maybe:
- either default tool detection to true, and define the tool detection to run in sync with the probe (optimally also use the detect state for internals -> like the verify maybe does?) or
- make it more intuitive to know that this isnt how we actually are detecting tools, despite it being called "detected tools", ie ranaming it to "probeless tool detecton" or something along the lines of that.
the verify is thus equally flawed in naming. what indicates that we couldnt verify our tool? that is being detected (its active, ie the probe is detected)

gcode offsets would likely be fixed by moving either:
- gcode offsets to variables (doability?)
- implementing a command to tells the toolchanger new values to use. 
so instead of [setting it from the config](https://github.com/viesturz/klipper-toolchanger/blob/e37603bfe7f355c8b825f3fe7d3733c754f74668/klipper/extras/toolchanger.py#L452 ), default to the config, and if not use those from a variable? or internal storage that can be set using something similar to TOOL_CALIBRATE_SAVE_TOOL_OFFSET 
something to note on that command aswell. it used to be present as an example in the configs, now it is nowhere to be found except in the actual source code. so either you look at the code, or hear from it by word of mouth (questionable)
[image]

last thing id like to add, ill add to the other issue request.

Contomo, 2025-03-26:
https://github.com/viesturz/klipper-toolchanger/issues/61#issuecomment-2754937220
Linking this here

Contomo, 2025-03-26:
```
 def select_tool(self, gcmd, tool, restore_axis):
        if self.status == STATUS_UNINITALIZED and self.initialize_on == INIT_FIRST_USE:
            self.initialize(self.detected_tool)
        if self.status != STATUS_READY:
            raise gcmd.error(
                "Cannot select tool, toolchanger status is %s" % (self.status,))

        if self.active_tool == tool:
            gcmd.respond_info(
                'Tool %s already selected' % tool.name if tool else None)
            return

        self.ensure_homed(gcmd)
        self.status = STATUS_CHANGING
        gcode_status = self.gcode_move.get_status()
        gcode_position = gcode_status['gcode_position'] #(1) current gcode pos?
        current_z_offset = gcode_status['homing_origin'][2]
        extra_z_offset = current_z_offset - (self.active_tool.gcode_z_offset if self.active_tool else 0.0)

        extra_context = {
            'dropoff_tool': self.active_tool.name if self.active_tool else None,
            'pickup_tool': tool.name if tool else None,
            'restore_position': self._position_with_tool_offset( #(2) restore pos = gcodepos + tool offsets? that means already applied offset + the offset (double?) ?
                gcode_position, restore_axis, tool, extra_z_offset),
            'start_position': self._position_with_tool_offset(
                gcode_position, 'xyz', tool, extra_z_offset)
        }

        self.gcode.run_script_from_command(
            "SAVE_GCODE_STATE NAME=_toolchange_state")

        before_change_gcode = self.active_tool.before_change_gcode if self.active_tool else self.default_before_change_gcode
        self.run_gcode('before_change_gcode', before_change_gcode, extra_context)
        self.gcode.run_script_from_command("SET_GCODE_OFFSET X=0.0 Y=0.0 Z=0.0") #(3) only now set offsets to 0 (should be done before getting gcode pos?)

        if self.active_tool:
            self.run_gcode('tool.dropoff_gcode',
                           self.active_tool.dropoff_gcode, extra_context)

        self._configure_toolhead_for_tool(tool)
        if tool is not None:
            self.run_gcode('tool.pickup_gcode',
                           tool.pickup_gcode, extra_context)
            if self.has_detection and self.verify_tool_pickup:
                self.validate_detected_tool(tool, gcmd)
            self.run_gcode('after_change_gcode',
                           tool.after_change_gcode, extra_context)

        self._restore_axis(gcode_position, restore_axis, tool) # (1.1) Restores the axis. Restore axis is currently split between user (pickup gcode) and also set internally no matter what user decides.

        self.gcode.run_script_from_command(
            "RESTORE_GCODE_STATE NAME=_toolchange_state MOVE=0") # (1.2) Restores the axis again. but now doesnt force move.
        # Restore state sets old gcode offsets, fix that.
        if tool is not None:
            self._set_tool_gcode_offset(tool, extra_z_offset)

        self.status = STATUS_READY
        if tool:
            gcmd.respond_info(
                'Selected tool %s (%s)' % (str(tool.tool_number), tool.name))
        else:
            gcmd.respond_info('Tool unselected')
```

Contomo, 2025-03-26:
#66

rbricheno, 2025-03-31:
There's another really good reason to want to modify the gcode offsets in a running system: if you try sensorless homing, you will have microstep errors (differences) each time you restart the printer.

Basically I want to start each print with a calibration where I (a) find the sensor, and then (b) the offsets, and then start printing with that sensor location and offsets. If I have to restart the printer between steps a and b, I have a good chance of introducing the micrtostepping error.

This might be as simple as adding a method like `set_offset(self):` in `Tool` and an option to call that in a new `cmd_TOOL_CALIBRATE_TOOL_OFFSET_LIVEUPDATE(self, gcmd):`

I can write that easily, but I wanted to check if there was a "gotcha" here, or a better way to do it.

Contomo, 2025-04-12:
> or is this called out by enough people
what defines as a callout by people. where would you be able to see or hear a callout?
> Documentation - happy to accept pull requests that update the documentation.
honestly.... it feels really weird for me to try to understand changes and document them if i havent made any of those changes myself.
reverse engineering someones code just to provide documentation seems like... a lot.

what would you think of moving the gcode offset handling into the macros?
keeping the internals all in absolute, adding set gcode offset 0,0,0 in the before change gcode, and set gcode offset tool.gcode_?_offset in the pickup/restore section.
that way any restore commands would restore to the right positions, while keeping the offsets handling ambiguous.

something certainly has to change anyway as the current offset handling is just broken.

im willing to hear ideas, or work on implementations, however its hard for me to bring some solutions to the table when things tend to go dormant for a while.

Contomo, 2025-06-22:
> I was expecting people to develop their own macros for various configurations, which seems to not be the case.
is there a good place to submit those macros? perhaps in the usermods? or should they stay seperated in their own repos?

closed 2025-09-20 by Contomo