viesturz/klipper-toolchanger #65: [Feature Request] SELECT_TOOL_INTERRUPT

Contomo, 2025-03-23:
basically SELECT_TOOL_ERROR, except that it doesnt error klipper (why is this even there, can always call ABORT, does the same id think)
SELECT_TOOL_INTERRUPT would basically skip any further internal execution of the change code.

viesturz, 2025-03-23:
Can you elaborate a bit more on the use case?

What should happen to any commands that are called after T<N> or SELECT_TOOL?
What should be the status of the toolchanger afterwards? Unitialized?

Contomo, 2025-03-23:
i guess uninitialized would work? the basic ideas is to allow users to prevent crashing klipper on a failed toolchange. exiting out or avoiding the rest of the internals to be ran inside "select_tool" 
The way its done right now is moving all the gcode sections into your own macros, checking for a crash after verify. and only then allow executing the rest of the macros that would run after the "verify" point
thus allowing select tool to be ran without actually doing anything.
to basically "let it run" (prob the only way this works with klipper)

SELECT_TOOL_INTERRUPT would just set an "interrupted?" flag inside, skipping the further internals from being called, ie the after change section from running, the gcode offsets being changed etc. 

this however is problematic as the pickup gcode usually contains the restoration, so id like to propose the idea of splitting the pickup gcode section into a pickup and restore gcode section alongside that to allow users to implement their own ways to handle failed toolchanges.

this way the main structure wouldnt change except for the move of the whole rtounded G0 stuff one space down into the restore gcoder section, yet users could call their own checks and interrupt. (herein interrupting the select process, calling their own macro on what to do now, which could very well be just a "PAUSE" )

viesturz, 2025-03-31:
I'm still not fully there what is the use case. Pause the print on failed toolchange, allowing the operator to fix the mess and resume?

Just to set expectations, failed toolchange often comes with missed steps and requires re-homing.

Contomo, 2025-03-31:
i have not yet had a failed toolchange which had lost steps.
allow the operator to fix it yes.

imo this is a secondary request. merely an idea. the other issue post i have with the PR attatched is meant more as a genuine thing.

viesturz, 2025-04-12:
Hi, added this in https://github.com/viesturz/klipper-toolchanger/commit/985c9a7d354b38d29250f49ad2361d8d75d88e92
It' s not a SELECT_TOOL_INTERRUPT  but rather different behavior of SELECT_TOOL_ERROR and VERIFY_TOOL_DETECTED.
Let me know if this works for you

Contomo, 2025-04-13:
looks good for now, ill play around with it over the next days.
i personally have not yet used VERIFY_TOOL_DETECTED before as im still using the tool probe endstops.
unrelated to this (just tried to find info about it)
[the fixed dock example](https://github.com/viesturz/klipper-toolchanger/blob/dd54695346cc9fd8772a105ce4a17a3ad5de658d/examples/dock%20location/fixed/toolchanger.cfg#L87) is using both _verify... and verify... commands.

closed 2025-04-13 by Contomo