viesturz/klipper-toolchanger #69: Sanity check the restore position stuff

Contomo, 2025-04-05:
i think there has been a flaw in the restore position calculations for like the past 3 years now.
when was the last time someone did a "sanity check" on the calculations inside?

not only is `gcode_position = gcode_status['gcode_position']` grabbed with all the offsets, only for then to supply the macros with the 
`'restore_position': self._position_with_tool_offset(gcode_position, restore_axis, tool, extra_z_offset)`
position with all the new tools offsets just added upon that, and also the "extra_z_offset" ofc.
but also is  `self._restore_axis(gcode_position, restore_axis, tool)` missing the `extra_z_offset `

meaning that just in the macro code, and internal python code. we are already moving to two different places.

Everything about the current state of gcode offset handling is inconsistent and incoherent.
i mentioned this in a previous issue, maybe it wasnt seen with the amount of current open issues so im making one for it here.

Contomo, 2025-04-12:
**Lets assume**
<div class="tableContainer horzScrollShadows">
  <table class="min-w-full">
    <thead>
      <tr>
        <th>Tool</th>
        <th>X Offset</th>
        <th>Y Offset</th>
        <th>Z Offset</th>
      </tr>
    </thead>
    <tbody>
      <tr>
        <td><strong>Tool T1 (old)</strong></td>
        <td><strong>+10</strong></td>
        <td><strong>+10</strong></td>
        <td><strong>+10</strong></td>
      </tr>
      <tr>
        <td><strong>Tool T2 (new)</strong></td>
        <td><strong>−10</strong></td>
        <td><strong>−10</strong></td>
        <td><strong>−10</strong></td>
      </tr>
    </tbody>
  </table>
</div>

**and a starting position of:**
<div class="tableContainer horzScrollShadows">
  <table class="min-w-full">
    <thead>
      <tr>
        <th>Position</th>
        <th>X</th>
        <th>Y</th>
        <th>Z</th>
      </tr>
    </thead>
    <tbody>
      <tr>
        <td><strong>absolute</strong></td>
        <td><strong>110</strong></td>
        <td><strong>110</strong></td>
        <td><strong>10</strong></td>
      </tr>
      <tr>
        <td><strong>gcode</strong></td>
        <td><strong>100</strong></td>
        <td><strong>100</strong></td>
        <td><strong>10</strong></td>
      </tr>
    </tbody>
  </table>
</div>

`gcode_position = gcode_status['gcode_position']`
Retrieves our starting position in gcode space of 100,100,10
`current_z_offset = gcode_status['homing_origin'][2]`
gets 0
`extra_z_offset = current_z_offset - (self.active_tool.gcode_z_offset if self.active_tool else 0.0)`
=> -10 = 0 - 10

```
            'restore_position': self._position_with_tool_offset(
                gcode_position, restore_axis, tool, extra_z_offset),
```

```
            'start_position': self._position_with_tool_offset(
                gcode_position, 'xyz', tool, extra_z_offset)
```

<div class="tableContainer horzScrollShadows">
  <table class="min-w-full">
    <thead>
      <tr>
        <th>Context</th>
        <th>X</th>
        <th>Y</th>
        <th>Z</th>
      </tr>
    </thead>
    <tbody>
      <tr>
        <td><strong>restore_position</strong></td>
        <td><strong>90</strong></td>
        <td><strong>90</strong></td>
        <td><strong>-10</strong></td>
      </tr>
      <tr>
        <td><strong>start_position</strong></td>
        <td><strong>90</strong></td>
        <td><strong>90</strong></td>
        <td><strong>-10</strong></td>
      </tr>
    </tbody>
  </table>
</div>

> this is whats passed into our pickup gcode section,
meanin we are restoring to 
**90,90,-10 in absolute terms**
**100,100,0 in T1 nozzle terms**

`self.gcode.run_script_from_command("SET_GCODE_OFFSET X=0.0 Y=0.0 Z=0.0")`

`self._restore_axis(gcode_position, restore_axis, tool)`

<div class="tableContainer horzScrollShadows">
  <table class="min-w-full">
    <thead>
      <tr>
        <th>Axis</th>
        <th>G-code Pos</th>
        <th>New Tool Offset</th>
        <th>Final Restore Pos</th>
      </tr>
    </thead>
    <tbody>
      <tr>
        <td><strong>X</strong></td>
        <td>100</td>
        <td>−10</td>
        <td><strong>90</strong></td>
      </tr>
      <tr>
        <td><strong>Y</strong></td>
        <td>100</td>
        <td>−10</td>
        <td><strong>90</strong></td>
      </tr>
      <tr>
        <td><strong>Z</strong></td>
        <td>10</td>
        <td>−10</td>
        <td><strong>0</strong></td>
      </tr>
    </tbody>
  </table>
</div>

> **G0 executed with X90 Y90 Z0**
gcode pos is what we care abou tho, which is **100 100 10**
(still wrong, because wed like to be at **100 100 0** because of our global z offset of -10!)

`self._set_tool_gcode_offset(tool, extra_z_offset)`

<div class="tableContainer horzScrollShadows">
  <table class="min-w-full">
    <thead>
      <tr>
        <th>Axis</th>
        <th>G-code Pos</th>
        <th>New Tool Offset</th>
        <th>Absolute pos</th>
      </tr>
    </thead>
    <tbody>
      <tr>
        <td><strong>X</strong></td>
        <td>100</td>
        <td>−10</td>
        <td><strong>90</strong></td>
      </tr>
      <tr>
        <td><strong>Y</strong></td>
        <td>100</td>
        <td>−10</td>
        <td><strong>90</strong></td>
      </tr>
      <tr>
        <td><strong>Z</strong></td>
        <td>20</td>
        <td>−10 -10(extra)</td>
        <td><strong>0</strong></td>
      </tr>
    </tbody>
  </table>
</div>

viesturz, 2025-04-12:
I'm pretty sure calculations are not right.

> current_z_offset = gcode_status['homing_origin'][2]
> gets 0

Homing origin literally stores the gcode z offset; https://github.com/Klipper3d/klipper/blob/61c0c8d2ef40340781835dd53fb04cc7a454e37a/klippy/extras/gcode_move.py#L202

Contomo, 2025-04-12:
> I'm pretty sure calculations are not right.
> 
> > current_z_offset = gcode_status['homing_origin'][2]
> > gets 0
> 
> Homing origin literally stores the gcode z offset; https://github.com/Klipper3d/klipper/blob/61c0c8d2ef40340781835dd53fb04cc7a454e37a/klippy/extras/gcode_move.py#L202

yes thats correct. our current z offset retrieved from the homing origin would be 0.
since the tool has 10, and we baby stepped it down back to 0

Contomo, 2025-04-14:
even if it wasnt correct tho...
equations would still be the same. just coming out to the restore being wrong in whats passed to the user, and correct internally.

Contomo, 2025-05-23:
@viesturz is this going to be adressed at any point again? or closed if not planned/disregarded?

closed 2025-09-20 by Contomo