viesturz/klipper-toolchanger #70: STOP_TOOL_PROBE_CRASH_DETECTION uses print time instead of system time, crash on 2nd tool change

idonasch, 2025-04-11:
If I interpret the code correctly then STOP_TOOL_PROBE_CRASH_DETECTION should stop the crash detection when the current moves are finished. It uses the toolheads get_last_move_time() function which returns the time relative to the main MCU's start time. But the reactor register_callback function uses system time and not print time. Usually the system time is ahead of the print time which can cause a crash detection at the time the tool is being released in the bay. This only happens right after startup if the toolheads MCU  has not yet detected a status change.
Anyways the implementation seems to be wrong and can be replaced with ether:

  ```
  def cmd_STOP_TOOL_PROBE_CRASH_DETECTION(self, gcmd):
        self.toolhead.wait_moves()      # M400
        self.stop_crash_detection()
```

or in the more fancy way of using the print time: 

 ```
   def cmd_STOP_TOOL_PROBE_CRASH_DETECTION(self, gcmd):
         self.toolhead.register_lookahead_callback(self.stop_crash_detection)

         def stop_crash_detection(self, _): # _ is the print time (not being used here) and only useful if executed on the MCU, usually it's executed right away.
                self.crash_lasttime = 0.
                self.crash_detection_active = False
```

Both seem to work for me. (verified with debug output)

viesturz, 2025-04-12:
You are correct.  This needs to be the lookahead callback obviously.
In practice this stops the detection too early, so not a huge problem. Thanks for spotting.

closed 2025-04-12 by viesturz

idonasch, 2025-04-13:
`self.toolhead = self.printer.lookup_object('toolhead')`
does not exist yet __init__()
Line in __init__ can be removed, it's already handled in _handle_connect()

commit 0f6c82cd (2025-04-12): Fix incorrect use of reactor time vs print time.

Fixes #70