viesturz/klipper-toolchanger #72: New tollchanger.cfg error

printicus, 2025-04-12:
I tryed the new toolchanger.cfg and got this error:

[image]

viesturz, 2025-04-12:
Hmm, interesting, I have a 30 hour print running (with a error_gcode in the config) so cannot double check right now.
Did you restart klipper after update? That should happen automatically if done via the UI.

printicus, 2025-04-12:
Actually never mind... I reverted to prevoius tollchanger.cfg. I see i got to upgrade to allot of files(manually, i dont have an automated check for update in moonraker) and for now my setup works as is... First i will backup what i have and then upgrade.

closed 2025-04-12 by printicus