viesturz/klipper-toolchanger #73: Unknown config object 'toolhead'

Contomo, 2025-04-12:
`  File "/home/contomo/klipper/klippy/extras/tool_probe_endstop.py", line 15, in __init__
    self.toolhead = self.printer.lookup_object('toolhead')`
with edit [985c9a7](https://github.com/viesturz/klipper-toolchanger/commit/985c9a7d354b38d29250f49ad2361d8d75d88e92 )

Contomo, 2025-04-12:
https://github.com/viesturz/klipper-toolchanger/pull/74#issue-2990813085

printicus, 2025-04-13:
Same error here. I reinstalled klipper-toolchanger, updated all the files in the example folder that are relevant to my setup(fixed docks, per tool probe) and got this... 

[image]

[klippy.log](https://github.com/user-attachments/files/19724591/klippy.log)

printicus, 2025-04-13:
I've kinda made it work... I dont know if I did good, but thi is what i did:

I comented out that line in tool_probe_endstop.py, it worked but getting this error: Unknown command:"_VERIFY_TOOL_DETECTED"
I looked and saw that the new macro, in tool_detection.cfg, is without the first underline, corected that macro call in toolchanger.cfg
in {% for pos in path %} section and now it works. At least for ramdom tool change. Not tested while printing.

commit 087f70f5 (2025-04-13): Fix too early toolhead lookup.
That's what you get when not testing :/

Fixes #73
commit d1807506 (2025-04-18): Yet another attempt to correct the tool validation scheduling.
Now tested in both sync and async nodes.

Fixes #73
closed 2025-04-13 by viesturz