viesturz/klipper-toolchanger #75: Next step in my build

printicus, 2025-04-13:
Hi
I am at the point where my setup works as good as it should. But i want more...(dont whe all?)
My ideea, from the begining, was to not use tap as z probing metod, but started with tap just to make all work. Now i want to switch to klicky-probe. For me this is a much better option and i like klicky allot.
I added [include klicky-probe.cfg] to printer.cfg, comented out the [homing-override] part from klipper-toolchanger homing.cfg so that homing would be executed from klicky's homing-override. I added INITIALIZE_TOOLCHANGER at the beginning and the other two macros at the end and tryed to home. Apparently, some klicky code still looks after [probe] section, which is no more, being replacet by [tool_probe Tn].
This are the file I modified:

[image]

[homing.cfg.txt](https://github.com/user-attachments/files/19726100/homing.cfg.txt)
[klicky-macros.cfg.txt](https://github.com/user-attachments/files/19726098/klicky-macros.cfg.txt)
[klippy.log](https://github.com/user-attachments/files/19726099/klippy.log)

Can I get some opinions, help...?

Contomo, 2025-04-20:
this just means you didnt define the probe in your user variables.
you can prob just add probe: "tool probe T0" or something. or probe: "virtual endstop z" or something, dont know what klicky requires.

printicus, 2025-05-05:
> this just means you didnt define the probe in your user variables. you can prob just add probe: "tool probe T0" or something. or probe: "virtual endstop z" or something, dont know what klicky requires.

Yes, you are right. It is in the klicky-variable.cfg and it looks like that:

```
gcode:
    {% set tool = printer.toolchanger.tool %}
    {% set Mx = printer['configfile'].config["stepper_x"]["position_max"]|float %}
    {% set My = printer['configfile'].config["stepper_y"]["position_max"]|float %}
    {% set Ox = printer['configfile'].config["probe"]["x_offset"]|float %}
    {% set Oy = printer['configfile'].config["probe"]["y_offset"]|float %}
    {% set Oz = printer['configfile'].config["probe"]["z_offset"]|float %} 
```

And I changet like that:

```
gcode:
    {% set Mx = printer['configfile'].config["stepper_x"]["position_max"]|float %}
    {% set My = printer['configfile'].config["stepper_y"]["position_max"]|float %}
    {% set Ox = 0.0 %}
    {% set Oy = 19.75 %}
    {% set Oz = 0.0 %}
```
I would like to link the Ox, Oy and Oz to the active tool but I did not succeded. They are used to set the probe to nozzle offset.

Now it works ok. 
Till I want to do a bed mesh.... And it looks like that:

[image]

Looks like somwhere its added twice the  z-offset. 
My current z-offset is -8.790 with klicky. I tried some bogus z-offsets like -5 or 5 just to be a round number and the rezult reflect that. If I put z-offset 0 the hight map is ok.

[image]

Anny ideas?

My files:

[klippy (with -8.790 z_offset.)log.log](https://github.com/user-attachments/files/20034991/klippy.with.-8.790.z_offset.log.log)

[klippy (with 0 z_offset.)log.log](https://github.com/user-attachments/files/20034994/klippy.with.0.z_offset.log.log)

[config-20250505-125231.zip](https://github.com/user-attachments/files/20035082/config-20250505-125231.zip)

I modified:
 klipper-toolchander homing .cfg:
           commented out the homing-override, and modified the _ADJUST_Z_HOME_FOR_TOOL_OFFSET macro so that it applies 
            correct the z_offset for klicky-probe. may need further modification, but for now i want to make it work with one tool.
klycki-variables.cfg like I have shown furter up.
klicki-macros.cfg to ad the klipper-toolchanger stuff to its homing-override section ( INITIALIZE_TOOLCHANGER, _ADJUST_Z_HOME_FOR_TOOL_OFFSET and _APPLY_ACTIVE_TOOL_GCODE_OFFSETS with M400).

printicus, 2025-05-10:
Solved it... kinda. Aparently the z-offset is not a "probe to nozzle" offset. Its a "nozzle to probe" offset. After babysteping to find out the z-offset value you must change te "polarity" of that value to work... like from negative to pozitiv. My babystepping value was -8.790mm, but to work I needed to put in the tool config z-offset section 8.790mm... contraintuitive. That happens whit TAP values allso. Babystepping a TAP z-offset brings out a positive value, but to work it must be put in tool config asa a negative value....

closed 2025-05-11 by printicus