viesturz/klipper-toolchanger #76: "Fix too early toolhead lookup" not yet ok?

printicus, 2025-04-13:
At print end i have a call for T0, because for now i can home only with T0. Before update i had no problem, now....

[image]

Contomo, 2025-04-14:
try adding an M400 before the verify, or a G4 P500

printicus, 2025-04-14:
Yeah... but I see even when I do random toolchange the activating next tool and next extruder and verifying the next tool apear in console before even droping off the curent tool, somewhere between safe y and close y. Maybe they are triggered by the wrong event... instead of the end of the pickup sequence, they are triggered by the begining of the dropoff sequence....

Contomo, 2025-04-14:
i really dont have enoug information to start looking into this. i looked at the log files yesterday and they diddnt really tell me much.

printicus, 2025-04-14:
No problem... I reverted to the version I had before upgrade....

viesturz, 2025-04-20:
Hi, I did another fix. The two different times in Klipper are really annotying. If you can update and let me know if it works now or not.

printicus, 2025-04-21:
Will do. Now I am in vacation... next week I will be home. Do I need to upgrade klipper too? I heard it comes with some errors.

printicus, 2025-04-27:
Updated and got this:

[image]

Not updated klipper. Will that make a difference?

Contomo, 2025-04-29:
Hard to tell as noone can see your config. it most likely sounds like you configured your tool probes very wrong.
but honestly, with the info you gave, no clue.

printicus, 2025-04-29:
OK, fair enough. Here they are:

[homing.cfg.txt](https://github.com/user-attachments/files/19966052/homing.cfg.txt)
[printer.cfg.txt](https://github.com/user-attachments/files/19966051/printer.cfg.txt)
[tool_detection.cfg.txt](https://github.com/user-attachments/files/19966053/tool_detection.cfg.txt)
[toolchanger.cfg.txt](https://github.com/user-attachments/files/19966056/toolchanger.cfg.txt)
[toolchanger-macros.cfg.txt](https://github.com/user-attachments/files/19966054/toolchanger-macros.cfg.txt)
[Toolhead_T0.cfg.txt](https://github.com/user-attachments/files/19966055/Toolhead_T0.cfg.txt)

Contomo, 2025-04-29:
so you only have one tool?

printicus, 2025-04-29:
I see you want to copy my configs.... :) I added the other tools to.
Actually I have three, for now. I say they are ok configured. They work when I do random toolchange, they work when printing. Only issue I have is when printing the first toolchange gives me that error. All other toolchanges are ok.

Contomo, 2025-04-29:
>  it most likely sounds like you configured your tool probes very wrong.

was why i asked for multiple tool configs.

various questionable things i have found in the configs you sent me 
you always call `VERIFY_TOOL_DETECTED` inside of your toolchanger pickup gcodes, while this would be okay i guess, 
you have a gcode macro 
```[gcode_macro VERIFY_TOOL_DETECTED]
rename_existing: VERIFY_TOOL_DETECTED_ORIG
    G4 P200
    DETECT_ACTIVE_TOOL_PROBE
    _STOP_IF_INCORRECT_TOOL {rawparams}```
overwriting the internal one? and instead doing a DIY detection using tool probe.
so i assume you are doing tool detection via the classic tap probe?
but yet you never `STOP_TOOL_PROBE_CRASH_DETECTION` anywhere, neither start it.
there is another tool detection pin configured in every tool, is that... the same as tool probes pin then?
```require_tool_present: True``` you defined this inside toolchanger yet never do anything with the tool detection?

further on, each tool has a duplicate
`[input_shaper]`, `[resonance_tester]` and `[adxl345]` section....
yet inside the actual tool object all the ```
#params_input_shaper_type_x: 'zv'
#params_input_shaper_freq_x: 52
#params_input_shaper_type_y: 'zv'
#params_input_shaper_freq_y: 37.2
```
are commented out?

im sorry there is so much going on in that config it may be better starting fresh....

printicus, 2025-04-30:
Ok... in the tool configs i will delete these sections but all other things, like the macro you pointed out are exactly how it is from viesturz... the duplicate call for that macro is a fix that i tried from one of the other issues from that repository and is the only modification regarding tool detection. I will delete it, but I started whitout that and the result was the same. The detection pin used in tool configs is different from the probe pin. I used the pin from the filament detection. Again, I commented out that also and did not make a difference. The lack of START_TOOL_PROBE_CRASH_DETECTION and STOP_TOOL_PROBE_CRASH_DETECTION is exactly how viesturz made the configs in the latest update... I did not modifyed annything. It is one commented out in the homing section, but its becouse I copied the homing part from the old version of homing.cfg witch i modified to my likeing, and the new version has no tool crash detection macro there.

printicus, 2025-04-30:
So.... Looking at my last answer, and your(Contomo) question above, I solvet the issue... Im so mad.... It is a START_TOOL_PROBE_CRASH_DETECTION macro that is not used when random changing tools, but is used when printing... in the Print_Start g-code in the printer.cfg. It it used before calling the prime_lines.cfg. I commented out that macro for start probe crash detection and in the two tests I made i had no errors.... F******CK. So much stress for this. Thank You all, and sorry for the headache...

closed 2025-05-03 by printicus