viesturz/klipper-toolchanger #77: update to klipper causes error

Marshalldog, 2025-04-19:
```
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 130, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 123, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 112, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/tool_probe_endstop.py", line 242, in load_config
    return ToolProbeEndstop(config)
  File "/home/pi/klipper/klippy/extras/tool_probe_endstop.py", line 24, in __init__
    self.homing_helper = probe.HomingViaProbeHelper(config, self.mcu_probe)
TypeError: __init__() missing 1 required positional argument: 'param_helper'
webhooks client 4134415160: New connection
webhooks client 4134415160: Client info {'program': 'Moonraker', 'version': 'v0.9.3-72-g7cdcca3'}
```
looked into this, error is caused by line 24 of tool_probe_endstop.py
the new klipper update added a new update to the HomingViaProbeHelper that added new arguments to the function, see: 
https://github.com/Klipper3d/klipper/blob/master/klippy/extras/probe.py

Raincad, 2025-04-19:
I am having same issue.

MacBoyPro98, 2025-04-19:
I've been looking into this and was able to resolve the problem. You can see the changes in [my PR](https://github.com/viesturz/klipper-toolchanger/pull/78)

closed 2025-04-20 by viesturz