viesturz/klipper-toolchanger #80: Internal error during connect: 'ProbeEndstopWrapper' object has no attribute 'probing_move'

Minihaugern, 2025-04-19:
got this error after update Internal error during connect: 'ProbeEndstopWrapper' object has no attribute 'probing_move'

Contomo, 2025-04-19:
please provide current klipper version and toolchanger version as there were a lot of hotfixes in the past day.

Minihaugern, 2025-04-19:
klipper
v0.13.0-50-gf2b27d17
klipper-toolchanger
v0.0.0-137-gdda9d284-inferred

Marshalldog, 2025-04-19:
same issue here

Contomo, 2025-04-20:
https://github.com/viesturz/klipper-toolchanger/pull/78#issuecomment-2816775113

MacBoyPro98, 2025-04-20:
Sorry everyone, I've been busy since my PR earlier. I'm working through this trying to debug

VIN-y, 2025-04-20:
This is the description from Klipper's commit:

```
Use the normal probe_session_start(), run_probe(),
pull_probed_results(), and end_probe_session() API from
ProbeSessionHelper.  This removes the custom probing_move() callback.
```

I kinna understand what they are doing; but, am a bit way out of depth to know how to fix it.

daTobi1, 2025-04-20:
Has the same problem.

VIN-y, 2025-04-20:
The commit that start breaking things for us is [Commit 6a87c5e](https://github.com/Klipper3d/klipper/commit/6a87c5e9f56d95b60ed6d7c471bfdc226353c97e).

For the mean time, you should rollback to Klipper v0.13.0-35 using KIAUH, while waiting for the issue to be fixed.

Minihaugern, 2025-04-20:
i have now rolled back v0.13.0-35 an get this error Internal error during connect: module 'extras.probe' has no attribute 'ProbeParameterHelper'

anyone got a solution?

daTobi1, 2025-04-20:
Same problem.

closed 2025-04-20 by viesturz

lodger666, 2025-04-20:
[image]

VIN-y, 2025-04-20:
yeah. I just notice that there is a recent update on the add-on side as well. which is no longer compatible with Klipper v0.13.0-35

Contomo, 2025-04-20:
i feel like we need a way for klipper toolchanger to automatically use the correct klipper version with each update ^^

daTobi1, 2025-04-20:
I have roll back to this Versions and this works.

klipper
 v0.13.0-35 > v0.13.0-50
klipper-toolchanger
 v0.0.0-134 > v0.0.0-139

Minihaugern, 2025-04-20:
what's the command to roll back klipper-toolchanger?

daTobi1, 2025-04-20:
cd ~/klipper-toolchanger
git fetch
git checkout d1807506

ceuqbf, 2025-04-27:
is there a command to roll back klipper to the correct versions

Minihaugern, 2025-05-04:
I just used kiauh to rollback klipper