viesturz/klipper-toolchanger #82: Problem after Klipper Update: Internal error during connect: __init__() missing 1 required positional argument: 'param_helper'

daTobi1, 2025-04-19:
After the Klipper update is get this error: Internal error during connect: __init__() missing 1 required positional argument: 'param_helper'

Log:
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/klippy.py", line 130, in _connect
    self._read_config()
  File "/home/biqu/klipper/klippy/klippy.py", line 123, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/biqu/klipper/klippy/klippy.py", line 112, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/biqu/klipper/klippy/extras/tool_probe_endstop.py", line 242, in load_config
    return ToolProbeEndstop(config)
  File "/home/biqu/klipper/klippy/extras/tool_probe_endstop.py", line 24, in __init__
    self.homing_helper = probe.HomingViaProbeHelper(config, self.mcu_probe)
TypeError: __init__() missing 1 required positional argument: 'param_helper'
webhooks client 281473032401968: New connection
webhooks client 281473032401968: Client info {'program': 'Moonraker', 'version': 'v0.9.3-72-g7cdcca3'}

Klipperversion: klipper v0.13.0-50-gf2b27d17
Toolchangerversion: klipper-toolchanger v0.0.0-134-gd1807506-inferred

Contomo, 2025-04-20:
Already fixed with [dda9d28](https://github.com/viesturz/klipper-toolchanger/commit/dda9d284c5b44491a99b11eb564b41a2ee5cfce5) youre just not running on the newest toolchanger version, but on the newest klipper version...

thatchriskid96, 2025-04-20:
Not sure if i should make a new issue or not, but even with the updated file i still receive this issue. 

Unhandled exception during connect
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/klippy.py", line 130, in _connect
    self._read_config()
  File "/home/biqu/klipper/klippy/klippy.py", line 123, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/biqu/klipper/klippy/klippy.py", line 112, in load_object
    self.objects[section] = init_func(config.getsection(section))
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/biqu/klipper/klippy/extras/tool_probe.py", line 173, in load_config_prefix
    return ToolProbe(config)
           ^^^^^^^^^^^^^^^^^
  File "/home/biqu/klipper/klippy/extras/tool_probe.py", line 26, in __init__
    self.endstop = self.printer.load_object(config, "tool_probe_endstop")
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/biqu/klipper/klippy/klippy.py", line 112, in load_object
    self.objects[section] = init_func(config.getsection(section))
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/biqu/klipper/klippy/extras/tool_probe_endstop.py", line 242, in load_config
    return 0.0
  File "/home/biqu/klipper/klippy/extras/tool_probe_endstop.py", line 24, in __init__
    self.param_helper = probe.ProbeParameterHelper(config)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: HomingViaProbeHelper.__init__() missing 1 required positional argument: 'param_helper'
webhooks client 281473122140048: New connection
webhooks client 281473122140048: Client info {'program': 'Moonraker', 'version': 'v0.9.3-72-g7cdcca3'}

viesturz, 2025-04-20:
Merged your pull request, thanks for fixing this!

closed 2025-04-20 by viesturz