viesturz/klipper-toolchanger #87: Probe not registering during homing

Car1derWa1, 2025-05-04:
i have a problem with the homing process. i have two printheads, each with a tap. i just want to use printhead 0 for homing. both sensors work, as i can see from the error messages on the console, e.g. when i trigger both. now the only problem is that the printhead rushes into the print bed without braking and without triggering the endstop.  virtual endstop is already set up. what should i do so that the software registers the sensor?

[calibrate-offsets.txt](https://github.com/user-attachments/files/20030139/calibrate-offsets.txt)
[Makros.txt](https://github.com/user-attachments/files/20030140/Makros.txt)
[T0.txt](https://github.com/user-attachments/files/20030137/T0.txt)
[T1.txt](https://github.com/user-attachments/files/20030138/T1.txt)
[Toolchanger.txt](https://github.com/user-attachments/files/20030141/Toolchanger.txt)

[printer.txt](https://github.com/user-attachments/files/20030142/printer.txt)

Car1derWa1, 2025-05-05:
I have now been able to narrow down the problem further. And come to the conclusion that there is a problem with the Toolprobe 0 and with the Detection Pin. If I set the detection pin to the same pin as the tool probe pin, I get an error [toolchanger failed to initialize, require_tool_present set and no tool present after initialisation] If I set the PIN to an unoccupied pin the probe Does the homing jusxt fine, But Klipper no longer recognises Which print head it uses when changing them.... any idea how to fix that. Or can someone please share his config.

Thanks in advance

printicus, 2025-05-05:
Dont use the detection pin in tool config. Just comment it out. It is usefull only if you have some kind of switch that triggers that pin(that must be other then the probe pin) when the tool is on the suttle.

closed 2025-05-05 by Car1derWa1