viesturz/klipper-toolchanger #88: Tool Probe and detection pin

printicus, 2025-05-08:
In my quest to adapt Klicky Probe to Klipper Toolchanger I come to a issue:
When starting the printer, or after a"firmware restart" for anny reason I get this:

[image]

I get allso error for not tool detected after a toolchange.
This is because the klicky probe is on the dock, and, beside using the detection pin, toolchanger software is stil looking for the probe.

Contomo, 2025-05-12:
well remove those calls then
remove the start tool probe detection, stuff. and just use tool detection?
enter the same probe for all the tools. (the error is likely that tool T0 doesnt have a tool probe defined)

printicus, 2025-05-12:
(Disclaimer: I sed in allot of posts that im not a software guy. I've started to learn, but i'm still far from knowing and understanding all... so in manny cases I will ask stupid question. So here whe go...)

What is that: "remove the start tool probe detection, stuff"? I've converted my Voron to Toolchanger with Klicky Probe, but that's 90% a mechanical change. If I shorten the Klicky magnets on the shuttle with a piece of megnetic metal, the printer works great in "TAP mode", the probe contacts role been taken by the pogopins. Then I dont have anny trouble whit printer starting, when doing an firmware restart or a toolchange. I'm using the detection pin for tooldetection(it can be seen in picture the toolchanger is initializet and the tool is active. So way does the KTC looks for probe? Yeah...
Please tell me what is ""remove the start tool probe detection, stuff".

Contomo, 2025-05-19:
it should work with tool too, if you provide the full name. ("tool T1") i think.

> "remove the start tool probe detection, stuff".

is referring to the fact that youre using tool detection with a single tool probe mounted to your shuttle.
this tool probe mounted to the shuttle can hardly ever "crash" 

> And because my probe (Klicky) is on the dock, no probes are open 

this entirely depends on the way you configured everything. is the klicky NO or NC, you mentioned pogopins (actually no clue what you meant with that)

>  If I shorten the Klicky magnets on the shuttle with a piece of megnetic metal, the printer works great in "TAP mode", the probe contacts role been taken by the pogopins

this would suggest that the behaviour is reversed. thats why i mentioned remove the tool probe crash detection. since. if the probe is not connected. or not triggered. a crash is assumed.
so when its not mounted. it is technically crashed.
you can of course use this feature to detect a dropped klicky. that would however need a different way of setup. im not sure if this is your goal, if it is, you need to think through the chain of wether to set a pullup or pulldown, where to invert or not to invert.

printicus, 2025-05-20:
` you mentioned pogopins (actually no clue what you meant with that)`

I sed I use klicky with a toolchanger. How do you think I connect the klicky probe that attaches on the shuttle with the wires coming from the toolhead? :)

[image]
[image]
[image]
[image]

`it should work with tool too, if you provide the full name. ("tool T1") i think`

Maybe, but I tryed like in the description and it did not work. I got it working like this: SET_ACTIVE_TOOL_PROBE T=0

Klicky is configured like on a regular, non toolchanger setup. Probe is NC. Just modified the klicky homing_overide do restore the offsets. In fact, now its all working like it should. I can print, change tool, all that it must do.

Only thing I want to resolve is having x-enstop on each tool. Physical switches, not sensorless. I have two solutions. Running two extra wires to each toolhead, connecting them in parallel and plugging the colective x-enstop in to the motherboard, or trying to make this work: https://github.com/viesturz/klipper-toolchanger/issues/35

Contomo, 2025-05-20:
> How do you think I connect the klicky probe that attaches on the shuttle with the wires coming from the toolhead? :)

umbilical too shuttle? anyways...
yeah the pictures paint a much clearer picture.

closed 2025-05-21 by printicus