viesturz/klipper-toolchanger #89: typo in github description

thijstriemstra, 2025-05-17:
it currently says "Toolcahnging extension for Klipper"

Contomo, 2025-05-19:
Its been there for 3 years. its staying.

BlackNet, 2025-05-19:
That's ok; everyone knows tool changer people can't spell ;)

thijstriemstra, 2025-05-20:
that's an unexpected response and quite frankly: idiotic. good luck with the project, im sure it's an extremely high quality project when simply typos like this are kept in there intentionally 😂

printicus, 2025-05-20:
> > Its been there for 3 years. its staying.
> 
> that's an unexpected response and quite frankly: idiotic. good luck with the project, im sure it's an extremely high quality project when simply typos like this are kept in there intentionally

You are good at making friends, and, I bet, fun at parties....

Contomo, 2025-05-20:
> > Its been there for 3 years. its staying.
> 
> that's an unexpected response and quite frankly: idiotic. good luck with the project, im sure it's an extremely high quality project when simply typos like this are kept in there intentionally

thank you very much for the reply. neither am i the owner of the project, nor do i have priviledges to edit it.
i thus cannot tell you if this has been kept intentionally. 
Im also not entirely certain if the quality of a project can be judged based merely on a typo in a mandatory github project description.

_on a finishing note, this is merely a github issue section, and not the comment section of stack overflow._

closed 2025-05-21 by thijstriemstra