viesturz/klipper-toolchanger #91: Probe triggered prior to movement (generic_cartesian)

mandela64, 2025-05-20:
Hello. I am using the new generic_cartesian kinematics and having the "Probe triggered prior to movement" after x+ probe. Not sure if the kinematics or an update caused the issue.

pwm=0.000
run_probe direction = z-
run_probe axis = 2, sense = -1

Probe made contact at 177.000000,312.000000,2.195625
run_probe direction = x+
run_probe axis = 0, sense = 1

[klippy-3.log](https://github.com/user-attachments/files/20345084/klippy-3.log)

Contomo, 2025-05-21:
first thigns first, tried querying the probe after that error to see if its actually triggered or not?

Contomo, 2025-05-21:
well :P 
that sounds more like a nudge issue.
are you probing in a heated chamber? non knurled screws?

mandela64, 2025-05-21:
Enclosed tridex and nudge has smooth copper screws. Chamber is cold during testing but I typically print abs. Also tested it using the gcode below and triggers fine manually.

[duplicate_pin_override]
pins: head1:PB12

[filament_switch_sensor head1_attached]
switch_pin: head1:PB12
pause_on_runout: false

Contomo, 2025-05-21:
heh! yeah. im also using copper screws.
i was experiencing similar issues, the way i resolved it was by using stronger magnets, or when running into that issue, flicking the probe around.
(idk might be ABS deposits or copper oxide/hydroxide/carbonate?)
Try it out to check if it fixes things, if it doesnt it may actually be because of config/tridex related things?

mandela64, 2025-05-21:
So after I comment out input shaper ... nudge starts working again. Was having issues before with my dual_carriage not homing because of input shaper and thought it was fixed with the patch. Guess not :(

Contomo, 2025-05-22:
aww wel.... i mean that sounds more like an issue with klipper. or something deeper in the source code.
il take a look at it when i find the time altho it does sound like its related to the dual carriage?

mandela64, 2025-06-25:
Found the solution `self.idex = config.has_section('dual_carriage') or config.has_section('dual_carriage u')
`

closed 2025-06-25 by mandela64