viesturz/klipper-toolchanger #94: Multiple toolchangers currently error on load.

Contomo, 2025-05-25:
> Config
[toolchanger]
Configures common tool changing parameters. More than one toolchanger can be configured, with arbitrary names. The unnamed toolchanger is always considered the main one and others can be connected to a tool in the main toolchanger.

"More than one toolchanger can be configured, with arbitrary names"
is stated in the main toolchanger readme file, yet, configuring two toolchangers will immediately lead to klipper erroring as the first thing it encounters is `self.gcode.register_command("INITIALIZE_TOOLCHANGER", self.cmd_INITIALIZE_TOOLCHANGER,` which already exists in our main toolchanger.

may the gcode command structure be changed to instead take an optional "toolchanger" parameter which rereoutes the commands? or rather, append the name of that toolchanger to the commands? ie:
```
        self.gcode.register_command("INITIALIZE_TOOLCHANGER",
                                    self.cmd_INITIALIZE_TOOLCHANGER,
                                    desc=self.cmd_INITIALIZE_TOOLCHANGER_help)
        self.gcode.register_command("SET_TOOL_TEMPERATURE",
                                    self.cmd_SET_TOOL_TEMPERATURE,
                                    desc=self.cmd_SET_TOOL_TEMPERATURE_help)
```
->
```
        def _cmd_name(self, base_cmd: str) -> str:
            suffix = f"_{self.name}" if self.name != "toolchanger" else ""
            return f"{base_cmd}{suffix}"

        self.gcode.register_command(self._cmd_name("INITIALIZE_TOOLCHANGER"),
                                    self.cmd_INITIALIZE_TOOLCHANGER,
                                    desc=self.cmd_INITIALIZE_TOOLCHANGER_help)
        self.gcode.register_command(self._cmd_name("SET_TOOL_TEMPERATURE"),
                                    self.cmd_SET_TOOL_TEMPERATURE,
                                    desc=self.cmd_SET_TOOL_TEMPERATURE_help)
... etc
```

ive not looked much more into the issue with multichangers as this already failing didnt give me the biggest confidence.

viesturz, 2025-06-08:
Hi, yes this was the original design but was never implemented.
I do not have a means of testing it right now.

Contomo, 2025-06-09:
> may the gcode command structure be changed to instead take an optional "toolchanger" parameter which rereoutes the commands? or rather, append the name of that toolchanger to the commands

pkease just pick one.

closed 2025-09-20 by Contomo