viesturz/klipper-toolchanger #95: Internal error on command:"TOOL_LOCATE_SENSOR" with newest Klipper version

Dan3dp, 2025-06-01:
Hi,

I'm experiencing something odd: after updating Klipper to the latest version (v13.0.124), I get the following error when trying to locate the sensor:

Internal error on command: "TOOL_LOCATE_SENSOR"

I have no idea which specific Klipper commit introduced the issue, so I rolled it back by 50 commits (to v13.0.74), and TOOL_LOCATE_SENSOR works as intended again.

What are your thoughts on this? Could you update 'tools_calibrate.py' to make it compatible with the latest Klipper version?

Thanks in advance for your reply!

Dan

viesturz, 2025-06-08:
Works fine here, v0.13.0-149-g607d0b42
Can you please attach your Klippy.log?

Dan3dp, 2025-06-08:
I again.

I have redone my configs using 'generic_cartesian' kinematics instead of 'cartesian' and used that specific type of definition for carriages and steppers and so on...

Now it also works on IDEX config without issues

Thanks for your time!

closed 2025-06-08 by Dan3dp

Dan3dp, 2025-06-09:
I have to reopen the issue....

After further testing, I realized that when probing from the X-axis direction with T1, I get the error: "probe triggered before movement."

I strongly believe this happens because, although the dual_carriage is moving, the script is still monitoring the movement state of the first carriage.

When probing with T0, everything works fine.

fsironman, 2025-06-19:
I created a pull request that fixed the problem on my Tridex: https://github.com/viesturz/klipper-toolchanger/pull/100